motioneye-project/motioneye #513: accessing motioneye from different network

jethroww, 2017-04-17:
is it possible to access motioneye from different network? like making the domain name public?

ccrisan, 2017-04-17:
This is not something that has to do with motionEye. Accessing your computer on a certain port from the public Internet is achieved by making sure you have a *public* (routable) IP address and forwarding a port on your router to the local computer. In case your IP is not fixed (i.e. it's dynamic), you'd need to set up a Dynamic DNS service, such as noip.com. Again, this has nothing to do with motionEye.

closed 2017-04-17 by ccrisan

jethroww, 2017-04-17:
I mean, in order to access motion eye i have to type _*my_RPi_IP_address*:8765_ as the URL right? if a person from other network wants to access it too, is that possible? or atleast be able to view its snapshot URL? I know that typing the same URL won't work on different PC unless it's the same network as the raspberry pi.

jethroww, 2017-04-17:
I mean, in order to access motion eye i have to type
*my_RPi_IP_address:8765* as the URL right? if a person from other network
wants to access it too, is that possible? or atleast be able to view its
snapshot URL? I know that typing the same URL won't work on different PC
unless it's the same network as the raspberry pi.

ccrisan, 2017-04-17:
The answer still stands: this has nothing to do with motionEye. You'll need to internally forward port 8765 to your PC running motionEye, just like you would do if you had any other server running on that machine.

jethroww, 2017-04-18:
thank you very much for the answer