motioneye-project/motioneye #514: Amcrest IPM-741

sham2017, 2017-04-19:
What are the camera settings for an Amcrest IPM-741 network camera on Motion Eye? On Zoneminder (ver 1.29.0), the options that work are Source type=Ffmpeg, Source=rtsp://[user]:[password]@[lan ip address]:554/cam/realmonitor?channel=1&unicast=true&proto=Onvif, Remote Method=RTP/Unicast. On Motion Eye, I entered "rtsp://[lan ip address]:554" for the URL, and entered the user name and password. It accepted the camera addition, but it doesn't stream.

jferrer-vhm, 2017-05-19:
I got it working with a IP2M-844E with the following URL, the user/password setup with the Amcrest View app, and choosing the UDP stream. I recommend you try the simplest API call first, test the stream, and then try different options.

rtsp://[ip]:554/cam/realmonitor?channel=1&subtype=1

drbago, 2017-05-19:
I got it working with a IP2M-844E with the following URL, the user/password setup with the Amcrest View app, and choosing the UDP stream. I recommend you try the simplest API call first, test the stream, and then try different options.

rtsp://[ip]:554/cam/realmonitor?channel=1&subtype=1

FatespinnerX9, 2019-03-31:
@JCFerrerG i know this is an old post, however, i created a github account just so i could tell you thank you.  I've been fighting with this for a long time and you have saved me. seriously thank man.

norman00, 2020-10-02:
thanks a lot for your answer @JCFerrerG

berttejeda, 2020-12-02:
Thank you! @JCFerrerG, this helped me as well.

Here's my setup:
* Host OS: **Centos 7**
* Docker Version: **19.03.8**
* Amcrest Camera Model: **IP8M-2496EB** ([Amazon Link](https://www.amazon.com/gp/product/B07C7W91YR/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1))
* My **docker-compose.yml**
```
version: "3.5"
services:
  motioneye:
    image: ccrisan/motioneye:master-amd64  # Change to ccrisan/motioneye:master-armhf for ARM chips (Pi etc.)
    network_mode: host
    volumes:
      - /etc/motioneye:/etc/motioneye
      - /etc/localtime:/etc/localtime:ro
      - /opt/motioneye:/var/lib/motioneye
```
* Adjusted my firewall rules: `sudo firewall-cmd --zone=public --add-port=8081/udp && sudo firewall-cmd --reload`
* Fired up motioneye `docker-compose up -d`
* Added the rtsp stream: `rtsp://[ip]:554/cam/realmonitor?channel=1&subtype=0` (Chose UDP)

The above worked flawlessly!

JamBalaya56562, 2026-11-07:
Closing as answered: the RTSP URL in the comment above works for several Amcrest models, as the replies confirm.

closed 2026-11-07 by JamBalaya56562