motioneye-project/motioneye #515: No cameras available

snowman82, 2017-04-20:
Hi,

I'am trying to get my raspi cam to work with motioneye for a few days without success...
I know the camera is not broken because I have taken I few screenshot with raspistill and made a few movies...  This was working till I upgraded motion eye.. I had an old version (cant remember which one) and everything was working... I have /dev/video0/  and I have added the bcm2835-v4l2 to my /etc/modules... 

When I try to add a camera and choose local MPEG camera, dont get any cameras available..

I have a Raspiberri Pi model b+ running RASPBIAN JESSIE WITH PIXEL + motioneye

Thanks,
Nuno

ccrisan, 2017-04-21:
"local MPEG camera" is not an existent option. What is the actual camera type that you have chosen?

snowman82, 2017-04-21:
Sorry I mean Simple Mjpeg camera
I have the raspberry camera module.

ccrisan, 2017-04-21:
Use the Local Camera option. If you don't have it, make sure you have motion installed.

closed 2017-04-21 by ccrisan

snowman82, 2017-04-21:
I have follwed your tutorial and ran the motion installation... 
pi@RASPICAM01:~ $ sudo service motion status
● motion.service - LSB: Start Motion detection
   Loaded: loaded (/etc/init.d/motion)
   Active: active (exited) since Thu 2017-04-20 22:34:57 WEST; 10h ago

Apr 20 22:34:57 RASPICAM01 systemd[1]: Started LSB: Start Motion detection.
pi@RASPICAM01:~ $ ~

ccrisan, 2017-04-21:
My tutorial never says that the motion service be started separately. You should *NOT* enable the motion service, as it will be started automatically when needed, as needed, by motionEye.

snowman82, 2017-04-21:
Hi,

found what was wrong... I previsouly had installed motion for testing.. and then installed the motioneye...

So I did:
`sudo apt-get purge motion`

and then reinstalled motion as states in your tutorial:

`sudo wget https://github.com/Motion-Project/motion/releases/download/release-4.0.1/pi_jessie_motion_4.0.1-1_armhf.deb`

`sudo  dpkg -i pi_jessie_motion_4.0.1-1_armhf.deb`

Now it is detecting the local camera and it is up and running.

Thanks