motioneye-project/motioneye #516: issue: motioneye on Ubuntu with more than 1 usb cams

DerDonDon, 2017-04-21:
I have installed motioneye on a brand new ubuntu on a x86 machine. 
I have attached 3 Micosoft usb cams - Microsoft LifeCam Cinema and 2 Microsoft LifeCam Studio.
No matter which USB Cam I pug in, and when or in which order I use Add Camera, always 
- I can add any cam any time
- the first added cam shows a picture
- the second or third added cam shows a grey picture with comment 
"UNABLE TO OPEN VIDEO DEVICE SINCE xxx-xx-xx xx:xx:xx"
and 
"Error capturing first date"

When I attach a pi with a cam or a Vimicro USB Cam, I can see the video streams.

Whats wrong ?

DerDonDon

amoulart, 2017-04-21:
Hello DerDonDon,

similar issue here on a rasp2 with 2 lifecam 3000hd.
Working perfect with one but can't make it work with 2.
I first though it was that the rasp not powerful enough but I don't known yet how to review the log.
I'll share my progress if any.

Alex

chrisf4lc0n, 2017-04-21:
You might be reaching the maximum bandwidth of the USB ports... Try to reduce the resolution.

amoulart, 2017-04-22:
Hello Chris,
you are right, reducing very very significantly the resolution solves the issue.
The rasp cannot handle 2 usb HD webcams + usb wifi...
Next step: 1 rasp per webcam

Thank you for your help.

DerDonDon, 2017-04-24:
I run a Core2Duo based Ubuntu - USB Port bandwith should not be an issue..... .

chrisf4lc0n, 2017-04-24:
The power of the CPU has got nothing to do with the USB bandwidth. It depends on how has been the motherboard designed, chipset etc.

MacAre, 2017-04-26:
To check if this is a usb port problem, try using another program (example: yawcam).
I had the same problem with 3 USB 320x240 webcam on PI3. I have disabled one, it works correctly with 2 webcam only (320x240). I have not tested with yawcam ...

DerDonDon, 2017-04-27:
chrisf4lc0n, MacAre you have fair points. But two or three HD720p cam should not be an issue for a PC - different to a Pi1, 2 or 3. 
I'd rather speculate about a driver issue. Does anyone know a way to test the cams 
- outside of motioneye
- outside of motion ?

chrisf4lc0n, 2017-04-27:
Since you have potentially brought up the drivers issue. I have to worry you a bit more. The problem might be a bit more complicated, not just the camera drivers play part here, if the chipset is not well supported by the OS/kernel you are using then you might be having the similar problem. On top of that it might the combination of the 2 (chipset, camera drivers) creating problems.
I would most probably try a different hardware combination if possible and then report it back.

JamBalaya56562, 2026-11-09:
Several cameras on one USB controller often run out of USB bandwidth, and lowering the resolution fixed it for amoulart. It is a limit of the hardware and driver rather than of motionEye, and there was no news since 2017, so closing.

closed 2026-11-09 by JamBalaya56562