motioneye-project/motioneye #517: zoom

Binog, 2017-04-23:
Hey there, 
is there a way to zoom into a camera image? I don't find an option in motioned.any other way to realize that ?

ccrisan, 2017-04-24:
There's no such option. If your camera supports some custom commands to zoom in/out, I'd recommend using [Action Buttons](https://github.com/ccrisan/motioneyeos/wiki/Action-Buttons) to control it.

closed 2017-04-24 by ccrisan

elmicha, 2018-12-04:
In Chrome Beta 71 on Android I can zoom in with the zoom finger gesture (it zooms the whole page). 

In desktop browsers zooming works only for a split second, then the image/video is resized to the browser window again.

Maybe there could be an option to turn that resizing off?

desegura, 2021-09-04:
@ccrisan It would be nice to have this option for MJPEG cameras, just under VideoDevice > Video Rotation

setting a scale (eg: 125%) and an X and Y deviation (eg: X:-4% & Y:+12%)