motioneye-project/motioneye #518: SSL certificate problem: self signed certificate

EpicGamer1YT, 2017-04-23:
I am trying to add a network camera. I have all of the credentials right, but it is giving me the error SSL certificate problem: self signed certificate. Is there any way around it without creating a self signed certificate? Thanks!

ccrisan, 2017-04-24:
I'll look into it.

EpicGamer1YT, 2017-04-24:
Is it possible to allow self signed certificates in motioneye.conf?

EpicGamer1YT, 2017-04-28:
@ccrisan is there anything to be done, because I have bought a network camera to connect. But it will not let me.

ccrisan, 2017-04-29:
There's something I need to change in the code for that to work, I think.

EpicGamer1YT, 2017-04-29:
@ccrisan Ok please message me when it's done. Thank You!

EpicGamer1YT, 2017-05-06:
@ccrisan did u get it working yet? I am waiting to link that camera to the system. Thanks!

ccrisan, 2017-05-28:
@EpicGamer1YT you'll just have to set [validate_certs](https://github.com/ccrisan/motioneye/wiki/Configuration-File#validate_certs) to `false` in your motioneye.conf.

closed 2017-05-28 by ccrisan

EpicGamer1YT, 2017-05-29:
@ccrisan It now shows not a supported network camera. My camera is a samsung ip camera running off of ethernet. What is wrong?

ccrisan, 2017-05-29:
@EpicGamer1YT you're not supplying a MJPEG stream URL, but probably the URL to the web frontend.

EpicGamer1YT, 2017-05-29:
@ccrisan how can I find the MJPEG stream of a samsung SNH-P6410BN

EpicGamer1YT, 2017-05-29:
@ccrisan nvm. I figured it out and connected it as an rtsp camera. Thanks!

diegodi33, 2019-04-22:
hi im using a foscam fi9826p on motion eye but i get a SSL certificate problem: self signed certificate on it 
how do i canect also without that i am using a pi3 whit motion eye plz help

diegodi33, 2019-04-22:
an i am using a ip

Sebyw931, 2021-01-07:
@ccrisan  I'm new here on the platform and would like to use Motion Eye for my IO broker. The installation has worked so far. But now I have the problem that the camera outputs the message "SSL certificate problem: unable to get local issuer certificate" when registering. As a beginner, can you explain to me how I set validate_certs to false? I am really at a loss.

starbasessd, 2021-01-07:
add line
validate_certs false
to the bottom of
/etc/motioneye/motioneye.conf
(theoretically)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?