motioneye-project/motioneye #519: camera pi V2 noir

pierrotori, 2017-04-25:
Hello
I use the motioneye with a camera PI V2 noir and i find the best configuration to have a good detection, i use only the **motion detection** not movies
i put  for the **preferences**
layout columns = 3
Layout rows = 1
the **resolution** = 1280*720
Motion detection
Frame rate change threshold 7%
Light switch detection 0%
Frame rate = 30
Motion GAP = 30
Minimum motion frames = 10
captured before = 2
captured after = 1

i have a capture frame rate between 15 and 20 fps with the configuration

On the web site: [http://www.sony-semicon.co.jp/products_en/new_pro/april_2014/imx219_e.html](url) we can make for the Imaging Size and Frame Rate full FOV for 1280*720HD video and 180 frames/s, but i try and i don't have the higher frame rate :( 

is it a good configuration or i can put a higher resolution and the higher frame rate  ?

Regards
Pierrotori

ccrisan, 2017-04-25:
You won't get more than 20-25 fps whatever you do with motionEye. It's because of all the processing that's done in the CPU.

closed 2017-04-25 by ccrisan

pierrotori, 2017-04-25:
What is the setting with this camera that you advise me ?