motioneye-project/motioneye #520: Improving performance

YairMobile19, 2017-04-25:
Hello,
First of all, thank you for this great piece of software.
Regarding performance - it seems that the motion daemon keeps analyzing the video frames even after a motion event has been detected, and it does so for every single frame (presumable to know EXACTLY when the motion has ended).
Do you know if it is possible to change this behavior?
I would happily trade a small latency in detecting motion END in favor of a 20 FPS recordings on RPI.
Even a settings of a constant period of recording ( after a motion has been detected) would be great.
This way, the RPI will analyze video frames at a low frame rate (approx 5 FPS for 640x480 vid size). Once a motion event has been detected it will record X seconds of video at a high frame rate ( no motion  analysis ). Afterwards it will go back to analyzing mode. 

Thank you.

ilovelinux, 2019-06-20:
See https://github.com/ccrisan/motioneyeos/issues/1170 https://github.com/ccrisan/motioneyeos/issues/1322 https://github.com/ccrisan/motioneye/issues/1112 https://github.com/ccrisan/motioneyeos/issues/105 https://github.com/ccrisan/motioneyeos/issues/1512
Workaround, since Motion isn't supposed to record audio: https://www.lavrsen.dk/foswiki/bin/view/Motion/SoundAudioRecording

jasaw, 2019-06-21:
This is a motion software question. Please ask in [motion](https://github.com/Motion-Project/motion) project.

closed 2022-03-20 by MichaIng