motioneye-project/motioneye #524: Error capturing first image - after several hours of operation

the-muldoon, 2017-04-26:
motionEye Version	20170212
Motion Version 4.0.1+git37b3595
Raspberry Pi 2
USB Webcam - Logitech HD 720P

After several hours (not exactly how many), I get a grey screen where is says: Error capturing first image. Rebooting the Pi fixes it... I'll be happy to supply any log files or other information; just tell me what you need.

tomwoodhams, 2017-09-18:
I have this exact issue too, with pi zero and pi cam noir. Strangely glad to see someone wth the same issue and different hardware. Any ideas?

ccrisan, 2017-09-18:
Motion will display a gray screen when the camera it's using becomes unavailable for some reason. This is not a valid motionEye issue. I'd recommend to check the motion log file for errors and, if you find anything strange there, report the issue to motion.

Keep in mind though that cases like these are usually linked to a hardware problem or a faulty camera driver, in which case there's nothing the guys from motion could do.

closed 2017-09-18 by ccrisan