motioneye-project/motioneye #525: Hi Calin. I cannot seem to get motioneye to capture video after motion is detected.  A folder is created in /var/lib/motioneye but i cannot seem to get video to save.. i thought it was a permission problem so i changed the folder permission with chmod 777 yet that did not fix the issue. Any help would be appreciated as i need to have this camera up and running asap due to security issues in my neighbourhood. .

Osirus126, 2017-04-26:


Osirus126, 2017-04-26:
Hi Calin. I cannot seem to get motioneye to capture video after motion is detected.  A folder is created in /var/lib/motioneye but i cannot seem to get video to save.. i thought it was a permission problem so i changed the folder permission with chmod 777 yet that did not fix the issue. Any help would be appreciated as i need to have this camera up and running asap due to security issues in my neighbourhood. .

ccrisan, 2017-04-26:
Please post your configuration (Movies, Motion Detection). Screenshots from the settings panel are fine.

Osirus126, 2017-04-27:
here is some info. 
[motion.log](https://pastebin.com/GvM8T4vC)

as you can see, my log says that mp4 is not supported. I think issue #490 is my answer. My issue is identical to Issue #875 which is closed now, but also refers to issue #490.

closed 2017-04-27 by Osirus126