motioneye-project/motioneye #526: Can't connect to wifi

donde321, 2017-05-01:
Downloaded latest MotionEyeOS for regular Pi. Extracted and burned to micro-sd for my Raspberry Pi Zero. I have wifi dongle and camera attached. Applied power. Using HDMI monitor. Boot starts. For some reason it can't find my wifi dongle. Reboots over and over Same problem. Using WPA/WPA2 option on router. Why does it look for wired network first? I look at other comments. Sort of similar. I was ready to check out MotionEyeOS for1st time. It's been around for a while and I hear very popular. Can't even get to started!

AshrafRezk, 2017-05-02:
hello @donde321 ,
Firstly, you need to make sure that you have a compatible dongle. what dongle do you have?
Secondly, the dongle will act as a client NOT an access point... which is the origin of your problem.
You need to configure it by entering your existent AP/router SSID and password; you must connect it via RJ for primary configuration.
[image]

JamBalaya56562, 2026-11-07:
Closing as answered above: the dongle has to be supported, and motionEyeOS needs your router's SSID and password, set up over a wired connection first. motionEyeOS is maintained in its own repository (motioneye-project/motioneyeos), so please ask there if this still happens.

closed 2026-11-07 by JamBalaya56562