motioneye-project/motioneye #529: Motion detection (Parking IOT)

AshrafRezk, 2017-05-02:
1- How do I adjust the motion detection sensitivity to send out only one picture/only one 4 second video.
2- How can I crop parts of the camera?

bscuderi13, 2017-05-07:
The ring doorbell pro detects motion from frame analyzing and then starts a recording for 30 seconds. I Would assume they shutdown the motion detection to get the better performance and just focus on the recording for the next 30 seconds. Though this has other disadvantages I would take that trade off if it could be accomplished. I just got an odroid xu4 and the performance is quite a bit better than the pi.

cyclops1982, 2017-05-31:
If i'm not mistaken, you can run a custom script. This script can take the image and movie as input parameters (%f i think).

JamBalaya56562, 2026-11-09:
The first part is there since 0.37: the capture mode "Motion Triggered (One Picture)" takes a single picture per motion event (#616). Cropping is not offered, but a motion mask limits detection to the parts of the picture you care about. Closing.

closed 2026-11-09 by JamBalaya56562