motioneye-project/motioneye #530: Impossible to install motioneye on RPi2

Kywalh, 2017-05-03:
Hello everyone !!!
I'm a newbie in programming and more particularly in Raspberry; even if I do have a solid HW designer background...
I have Jessie installed and did my first Python scripts...
I have also burn MotioneyeOS on another SD card and make it work perfectly !!!

The problem I have is that I would like to have everything on the same SD card and thus move to MotionEye and not MotioneyeOS...

I followed the tutorial here : https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
and I started to have issues when runnng the command:
`cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf`
where I had some `cp omitting directory` issue... solved by adding the -R...

My problem now is that when I run the command:
`sudo cp /usr/local/share/motioneye/extra/motioneye.systemd-unit-local /etc/systemd/system/motioneye.service`
I have a `cp: impossible to evaluate« /usr/local/share/motioneye/extra/motioneye.systemd-unit-local »: no file or directory of this type` (sorry, I have it in French so the translation might be wrong).

Of course, when I try to run the next commands in the tutorial, nothing works and I can't start motioneye...

Any idea ?

ccrisan, 2017-05-04:
Your errors don't make sense. The command `cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf` should not complain about any folder. I'd recommend making sure you don't have any extra spaces in that command and that you have typed it exactly like that.

Another idea is to remove the `local/` part from the commands. On some systems, the files are installed directly in `/usr/share`,  instead of `/usr/local/share`.

Kywalh, 2017-05-06:
Hello, 
I tried but unsuccessuflly ...
I resintalled Jessie from scratch and follow step by step with a lot of errors...

`pi@raspberrypi:~ $ sudo apt-get install python-pip python-dev curl libssl-dev libcurl4-openssl-dev libjpeg-dev libx264-142 libavcodec56 libavformat56 libmysqlclient18 libswscale3 libpq5
Reading package lists... Done
Building dependency tree       
Reading state information... Done
curl is already the newest version.
libx264-142 is already the newest version.
libx264-142 set to manually installed.
python-pip is already the newest version.
You might want to run 'apt-get -f install' to correct these:
The following packages have unmet dependencies:
 libavcodec56 : Depends: libavresample2 (>= 6:11~beta1) but it is not going to be installed
                Depends: libavutil54 (>= 6:11~beta1) but it is not going to be installed
 libavformat56 : Depends: libavutil54 (>= 6:11~beta1) but it is not going to be installed
 libjpeg-dev : Depends: libjpeg62-turbo-dev (>= 1:1.3.1-12) but it is not going to be installed
 libmysqlclient18 : Depends: mysql-common (>= 5.5.54-0+deb8u1) but it is not going to be installed
 libswscale3 : Depends: libavutil54 (>= 6:11~beta1) but it is not going to be installed
 motion : Depends: libavutil54 (>= 6:11~beta1) but it is not going to be installed
 python-dev : Depends: libpython-dev (= 2.7.9-1) but it is not going to be installed
              Depends: python2.7-dev (>= 2.7.9-1~) but it is not going to be installed
E: Unmet dependencies. Try 'apt-get -f install' with no packages (or specify a solution).
pi@raspberrypi:~ $ sudo apt-get install python-pip python-dev curl libssl-dev libcurl4-openssl-dev libjpeg-dev libx264-142 libavcodec56 libavformat56 libmysqlclient18 libswscale3 libpq5
Reading package lists... Done
Building dependency tree       
Reading state information... Done
curl is already the newest version.
libx264-142 is already the newest version.
libx264-142 set to manually installed.
python-pip is already the newest version.
You might want to run 'apt-get -f install' to correct these:
The following packages have unmet dependencies:
 libavcodec56 : Depends: libavresample2 (>= 6:11~beta1) but it is not going to be installed
                Depends: libavutil54 (>= 6:11~beta1) but it is not going to be installed
 libavformat56 : Depends: libavutil54 (>= 6:11~beta1) but it is not going to be installed
 libjpeg-dev : Depends: libjpeg62-turbo-dev (>= 1:1.3.1-12) but it is not going to be installed
 libmysqlclient18 : Depends: mysql-common (>= 5.5.54-0+deb8u1) but it is not going to be installed
 libswscale3 : Depends: libavutil54 (>= 6:11~beta1) but it is not going to be installed
 motion : Depends: libavutil54 (>= 6:11~beta1) but it is not going to be installed
 python-dev : Depends: libpython-dev (= 2.7.9-1) but it is not going to be installed
              Depends: python2.7-dev (>= 2.7.9-1~) but it is not going to be installed
E: Unmet dependencies. Try 'apt-get -f install' with no packages (or specify a solution).
pi@raspberrypi:~ $ sudo dpkg -i pi_jessie_motion_4.0.1-1_armhf.deb(Reading database ... 112269 files and directories currently installed.)
Preparing to unpack pi_jessie_motion_4.0.1-1_armhf.deb ...
Unpacking motion (4.0.1-1) over (4.0.1-1) ...
dpkg: dependency problems prevent configuration of motion:
 motion depends on libavcodec56 (>= 6:11~beta1) | libavcodec-extra-56 (>= 6:11.7); however:
  Package libavcodec56:armhf is not installed.
  Package libavcodec-extra-56 is not installed.
 motion depends on libavformat56 (>= 6:11~beta1); however:
  Package libavformat56:armhf is not installed.
 motion depends on libavutil54 (>= 6:11~beta1); however:
  Package libavutil54:armhf is not installed.
 motion depends on libmysqlclient18 (>= 5.5.24+dfsg-1); however:
  Package libmysqlclient18 is not installed.
 motion depends on libpq5; however:
  Package libpq5 is not installed.
 motion depends on libswscale3 (>= 6:11~beta1); however:
  Package libswscale3:armhf is not installed.

dpkg: error processing package motion (--install):
 dependency problems - leaving unconfigured
Processing triggers for systemd (215-17+deb8u6) ...
Processing triggers for man-db (2.7.0.2-5) ...
Errors were encountered while processing:
 motion
pi@raspberrypi:~ $ sudo pip install motioneye
Downloading/unpacking motioneye
[... 28 lines cut ...]
  File "<string>", line 17, in <module>

  File "/tmp/pip-build-j00ioq/pycurl/setup.py", line 823, in <module>

    ext = get_extension(sys.argv, split_extension_source=split_extension_source)

  File "/tmp/pip-build-j00ioq/pycurl/setup.py", line 497, in get_extension

    ext_config = ExtensionConfiguration(argv)

  File "/tmp/pip-build-j00ioq/pycurl/setup.py", line 71, in __init__

    self.configure()

  File "/tmp/pip-build-j00ioq/pycurl/setup.py", line 107, in configure_unix

    raise ConfigurationError(msg)

__main__.ConfigurationError: Could not run curl-config: [Errno 2] No such file or directory

----------------------------------------
Cleaning up...
Command python setup.py egg_info failed with error code 1 in /tmp/pip-build-j00ioq/pycurl
Storing debug log for failure in /root/.pip/pip.log
`

Kywalh, 2017-05-06:
Problem solved !!!!
I using the `-f update` to update dependencies and followed the install procedure and it works !!!

Thanks a lot !!!

closed 2017-05-06 by Kywalh

zoldaten, 2017-06-13:
Could you help with installation ? On what step should be "-f update" ?