motioneye-project/motioneye #532: Run MotionEye as Root

snowman82, 2017-05-03:
Hi,

how does one run motionEye as root  to enable smb shares ? I need to save my motion pictures and movies elsewhere... 

thanks,
Nuno

ccrisan, 2017-05-04:
It runs as root by default if you follow the [installation instructions](https://github.com/ccrisan/motioneye/wiki/Installation).

closed 2017-05-04 by ccrisan

tnikiel82, 2017-12-15:
Hello, 
I installed everything with the instruction. 

But when I run smb_share on TRUE, systemn not running.

Why?
tnikiel

MarcoMirisola, 2018-01-27:
Hi, 
same here, set smb_share to true make the service not running.
Anyone can help?

MarcoMirisola, 2018-01-28:
I found the solution [here](https://github.com/ccrisan/motioneye/issues/244#issuecomment-322562566) you should install cifs-utils !

Wojcioo, 2020-05-20:
Hi
I've installed MotionEye on Debian using this instruction: https://github.com/ccrisan/motioneye/wiki/Install-On-Debian. I've changed smb_share on TRUE and system is not running. Then I've installed install cifs-utils, and MotionEye started but there is still no Network Share Option.