motioneye-project/motioneye #533: Odroid XU4 Open Media Vault

bscuderi13, 2017-05-07:
First off great work with the program it is great! So thanks you!

I just invested in an odroid and I am much more impressed with the speeds compared to the pi. I installed motioneyeos for the testing. However, I would like to run open media vault on the odroid as well with a couple of hard drives for a make shift NAS and put my investment to multiple uses.

I was abe to get plain motion eye running on a raspberry pi but when I try to follow the step by step guide for debian on the xu4 the motioneye server works but I'm not seeing any of the options for motion detection and video recording. I assume I'm doing something wrong but if you want to try and replicate I downloaded the open media vault image from the hard kernel website and ssh'd in and attempted to install motioneye. If I could get these both working I would be a happy camper. Thanks :)

ccrisan, 2017-05-07:
What did you choose as camera type when you added your camera?

bscuderi13, 2017-05-11:
I believe mjpeg that was the only choice... the other cameras are fast network cameras. The same settings worked on the pi just not odroid. I think something might be wrong with the installation of motion?

ccrisan, 2017-05-11:
What does `motion -h` say?

bscuderi13, 2017-05-11:
you know what... I fixed it I'm not sure why but for whatever reason i did an apt-get update and an apt-get upgrade rebooted and then started from the top and that seemed to solve the problem. Thanks for the help. I'm guessing the os on this image was just too old for these instructions to take.

closed 2017-05-11 by bscuderi13