motioneye-project/motioneye #535: how to run ? py script in motioneyeos ??

ronpa357, 2017-05-13:
How to run py script in motioneyeos ?? i place my py script in the data folder

Also ccrisan can i run sudo and install some module in motioneye ?  thank you so much :)

jatininfo, 2017-05-15:
you can give script as "python /home/pi/web-server/action.py" in Run A Command	ON	
Command box. this is working for me.

djixyacine, 2018-03-25:
@jatininfo if you don't mind me asking, what's the "actions" that you're able to make on cameras?

JamBalaya56562, 2026-11-07:
Closing as answered: as jatininfo said above, a script can be given in the "Run a command" field of the motion notifications, for example `python /path/to/script.py`. Installing Python modules on motionEyeOS is a motionEyeOS question, which is maintained in its own repository (motioneye-project/motioneyeos).

closed 2026-11-07 by JamBalaya56562