motioneye-project/motioneye #536: meyectl: ERROR: buffer is not large enough

ralphhughes, 2017-05-17:
A few days ago I installed motioneye following these instructions:
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian

This worked great and I configured Motioneye to save video clips of motion captured on the raspberry pi camera. My pi has since been restarted a couple of times and I notice that the front end now says "An error has occurred, refreshing recommended" and my /var/log/daemon.log file contains:

> May 17 11:52:01 rpi-server2 meyectl[339]: ERROR: buffer is not large enough
May 17 11:52:01 rpi-server2 meyectl[339]: Traceback (most recent call last):
May 17 11:52:01 rpi-server2 meyectl[339]: File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 1467, in _stack_context_handle_exception
May 17 11:52:01 rpi-server2 meyectl[339]: raise_exc_info((type, value, traceback))
May 17 11:52:01 rpi-server2 meyectl[339]: File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 1669, in wrapper
May 17 11:52:01 rpi-server2 meyectl[339]: result = method(self, *args, **kwargs)
May 17 11:52:01 rpi-server2 meyectl[339]: File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 247, in get
May 17 11:52:01 rpi-server2 meyectl[339]: self.list()
May 17 11:52:01 rpi-server2 meyectl[339]: File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 175, in wrapper
May 17 11:52:01 rpi-server2 meyectl[339]: return func(self, *args, **kwargs)
May 17 11:52:01 rpi-server2 meyectl[339]: File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 668, in list
May 17 11:52:01 rpi-server2 meyectl[339]: ui_config = config.motion_camera_dict_to_ui(local_config)
May 17 11:52:01 rpi-server2 meyectl[339]: File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 1339, in motion_camera_dict_to_ui
May 17 11:52:01 rpi-server2 meyectl[339]: ui['mask_lines'] = utils.parse_editable_mask_file(data['@id'], capture_width, capture_height)
May 17 11:52:01 rpi-server2 meyectl[339]: File "/usr/local/lib/python2.7/dist-packages/motioneye/utils.py", line 910, in parse_editable_mask_file
May 17 11:52:01 rpi-server2 meyectl[339]: pixels = list(im.getdata())
May 17 11:52:01 rpi-server2 meyectl[339]: File "/usr/local/lib/python2.7/dist-packages/PIL/Image.py", line 1165, in getdata
May 17 11:52:01 rpi-server2 meyectl[339]: self.load()
May 17 11:52:01 rpi-server2 meyectl[339]: File "/usr/local/lib/python2.7/dist-packages/PIL/ImageFile.py", line 179, in load
May 17 11:52:01 rpi-server2 meyectl[339]: self.map, self.size, decoder_name, extents, offset, args
May 17 11:52:01 rpi-server2 meyectl[339]: ValueError: buffer is not large enough
May 17 11:52:01 rpi-server2 meyectl[339]: ERROR: 500 GET /config/list/?_=1495018320759&_username=admin&_signature=7fe9a1b06448412327b90eccdb57473984e7363c ([removed]) 267.88ms

Any idea what I've done or how I can fix this?

martinchodev, 2017-05-19:
I have the same issue. I installed motionEye yesterday on a Raspberry Pi 3 B, it worked for a while, but today I am stuck with this issue.

On a (maybe) related note, the device ran our of space overnight. I have now cleaned up enough space but the same problem persists.

ccrisan, 2017-05-20:
Can you attach your mask_1.pgm file? I'll also need to know your camera resolution settings.

ralphhughes, 2017-05-22:
Have uploaded mask_1.pgm and thread-1.conf here:
[mask_1_pgm.zip](https://github.com/ccrisan/motioneye/files/1018096/mask_1_pgm.zip)

thread-1.conf seems to contain:
width 800
height 600
So I assume this was my last config as I can no longer access any settings via the front end. Also I notice in that config folder, motion.conf is 0 bytes and has an identical last modified time to mask_1.pgm, I assume that's normal to have an empty file there?

m00k12, 2018-11-19:
I've got something very similar in my motioneye.log after any sort of access attempt from the browser, which also lacks camera settings and tells me "An error has occurred, refreshing recommended":

```
2018-11-19 21:13:13: [motioneye]    ERROR: 500 GET /config/list/?_=1542661992518&_username=admin&_signature=d6f4e843e56c5b189ef62f0b259bcd5c848f6546 (192.168.1.149) 115.23ms
2018-11-19 21:13:19: [motioneye]    ERROR: buffer is not large enough
Traceback (most recent call last):
  File "usr/lib/python2.7/site-packages/tornado/web.py", line 1467, in _stack_context_handle_exception
  File "usr/lib/python2.7/site-packages/tornado/web.py", line 1669, in wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 275, in get
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 205, in wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 718, in list
  File "/usr/lib/python2.7/site-packages/motioneye/config.py", line 1433, in motion_camera_dict_to_ui
  File "/usr/lib/python2.7/site-packages/motioneye/utils.py", line 941, in parse_editable_mask_file
  File "mnt/output/raspberrypi/target/usr/lib/python2.7/site-packages/PIL/Image.py", line 1187, in getdata
  File "mnt/output/raspberrypi/target/usr/lib/python2.7/site-packages/PIL/ImageFile.py", line 179, in load
ValueError: buffer is not large enough
```

This happened just after my storage filled right up and I tried to change some settings.

After that I removed the mask_1.pgm, and everything spring to life again.  Retrieving and trying to open the mask in Gimp reports it is indeed corrupted.

I've attached the file (Github made me gzip it first) in case you can fix handling of a corrupted mask file at this point so that the consequence isn't quite so severe.

Otherwise investigating what happens when the filesystem is full, or preventing that happening, might be a way to stop the mask getting corrupted? 

[mask_1.pgm.gz](https://github.com/ccrisan/motioneye/files/2597103/mask_1.pgm.gz)

ccrisan, 2018-11-26:
@m00k12 thanks for providing all these details. Indeed, when the SD card becomes full, the behavior will be unpredictable, including files corruption like in your case.

I am planning on making motionEye more proactive when it comes to filling up storage space, but I haven't had too much time lately.

developerfromjokela, 2019-10-29:
I solved that issue by removing all mask files.

closed 2026-11-08 by Marijn0