motioneye-project/motioneye #538: Pixel change graph over time

tehKaiN, 2017-05-20:
Hi there,

AFAIK motion detection is done by comparing pixel count changes over time with percentage threshold configurable in camera settings.

It would be extremely cool if motioneye could measure pixel changes at given frame and place that info on daily graph, so one could easily cross-match with videos which levels are noise and which levels are true motion detections.

Also, current debug places changed pixel count at top-right counter, but settings are configurable per-percent. It would be more useful to have percentage printed on frames instead (or additionally, in brackets?).

ccrisan, 2017-05-20:
Unfortunately motion doesn't allow us to do any of the things that you proposed.

JamBalaya56562, 2026-11-09:
As ccrisan said, Motion does not give us the changed pixels of every frame, so there is nothing to draw a graph from; "Show Frame Changes" still shows the count on the picture. Closing as not planned.

closed 2026-11-09 by JamBalaya56562