motioneye-project/motioneye #539: Camera not adjusting to light conditions

jorisbp, 2017-05-22:
Hi,

I'm facing a problem with my 2 cameras: they do not adjust to light conditions. However, every time i restart motioneye (or just reboot my rpi0), the camera seems to adjust fine to the current light conditions. 

I could start a cron job that restarts motioneye a few times a day, but this not a elegant solution...

My question is: is this a problem of motion/motioneye/picamera? and do you have any suggestion how i could to solve this?

I'm using 2 different (one fisheye and one normal 5mp) picameras i ordered from China on two RPI Zero W's (connected to camera interface). 

Thanks in advance for your answer,

Joris

FaBjE, 2017-05-25:
I have a similar thing and I believe it is due to motion's interfacing with the RPI camera.
It is currently not able to set the exposure/white balance settings to the RPI camera like raspivid and raspistill can.

You can "force" this by using the new mmalcam_control_params config option for motion, but you need a motion with these options enabled at compile time.

JamBalaya56562, 2026-11-07:
Closing, as you got it working by streaming the camera through streameye and adding it as a network camera. As noted above, `mmalcam_control_params` is the other way to pass exposure settings to the camera.

closed 2026-11-07 by JamBalaya56562