motioneye-project/motioneye #540: Computer Build for MotionEye

schmeckles22, 2017-05-22:
Hello,

I am not having great luck with the RPi 3 as it just doesn't have the power required. I will buy a second hand PC and install Arch... I was wondering what sort of CPU is required for a decent FPS stream from a USB 720p Camera and IP 1080P camera (Settings can be changed on the IP Camera)

schmeckles22, 2017-05-23:
@ccrisan what would you say the recommended CPU would be? Would a i3 2nd Generation be powerful enough? I am looking at computers that are around £70 and they are 2.8GHz - 3.2GHz

ccrisan, 2017-05-24:
No idea. I have only run it on ARM devices and my own PC (core i7 3rd gen). Obviously the experience on my core i7 is far superior to that of ARMs.

schmeckles22, 2017-05-24:
Ahhh ok thank you. I'm having an issue with the Pi3, when it detects the motion it will freeze for about 10 seconds and that part of the recording is missing, I'm putting it down to the CPU not being good enough?

chrisf4lc0n, 2017-05-27:
What is your CPU utilisation on the current config?

fredrik-rambris, 2017-07-28:
Also note that the Pi throttles down when the CPU gets too hot. Adding cooling to the CPU so that it stays under 80 degrees C will keep it running at full speed.

mryogihbear, 2018-12-03:
Is it possible to run on a pc as a program i'm running Ubuntu 18.04

JamBalaya56562, 2026-11-09:
As answered, any recent x86 PC handles a couple of cameras far better than a Pi 3, and on a Pi, cooling keeps it from throttling. Closing, as the question is answered.

closed 2026-11-09 by JamBalaya56562