motioneye-project/motioneye #541: Feature/Request: multiple Surveillance user accounts

TyTnMonz, 2017-05-26:
Hi everybody, is it possible to create multiple "Surveillance user" ? I think this could be usefull for two main reasons: 
- the admin could know who is connected and, creating a log of the accesses, when
- could be created multiple profiles setting which camera every surveillor can manage on multiple cameras system (i.e.: I have 4 cameras and I wanna User1 can whatch only Cam1 and Cam2 while User2 can whatch only Cam3 and 4).

What do you think about this?

Thanks fro your works!
Best regards.

Noxigin, 2017-05-30:
Nice idea, would appreciate this.

STrRedWolf, 2020-09-15:
Seconded.  I want one user that can do a 2x2 display of four cameras, and one that does 3-across.  That way I can default to the 2x2 and I'll log into the 3-wide off of some displays (like my small downstairs bedroom TV).

SirPrikol, 2021-02-26:
+100

scan-dev, 2021-11-08:
would be nice additional feature indeed.

jul-fls, 2022-01-07:
Yes it would be very useful for the cctv system in my company, i need that the faster it's possible. Is there any workaound to get that until its possible ?

zagrim, 2022-01-07:
Multiple user accounts might (sort of) be possible to get by delegating user authentication to a proxy (nginx or apache), then the question of "who did what" should be possible to answer based on the proxy logs.
Different UI setting per user might be impossible without proper multi-user support in ME, at least if I remember correctly and the settings are only saved in the browser.

To be properly done this requires code changes. Maybe a more generic and more fine grained access control could be implemented in the same time, like giving one user access to only some of the cameras.

jul-fls, 2022-01-07:
Yes that's exactly what i want multiple user accounts, only 1 admin and for every account check which cameras it can view. Because in my company (high school) we want the head of training to be able to watch the cameras of his corner but not be able to watch others cameras so, it's complicated, how are user managed and stored in motioneye ? Because if it's json based or something like tthat i can edit the file to match what i want...

TyTnMonz, 2022-01-07:
I'm happy but also sad that my thread is still open after 5 years XD.  In the end I simply opted for the implementation of multiple instances of MotionEye by installing it locally on single Raspberry equipped with monitors and configuring them individually so as to be able to manage different cameras based on the computer used by user X rather than Y. The fact remains that it would undoubtedly be more convenient to be able to manage multiple logins and configure only those instead of the whole system, even if it is enough to make a clone of the SD and then configure only the webcams within the single instance.

Happy and healthy 2022 to you all.

jul-fls, 2022-01-07:
Hello, happy 2022 to you also.
I'm thinking maybe with a reverse proxy and a .htaccess i may be able to do something, idk if it's possible to implement that easily, also do you know if it's possible to block access to viewing cameras if the user is guest (not authenticated) because if you go to the right ip and port it doesn't ask credentials by default and displays cameras which is not very safe.

TyTnMonz, 2022-01-07:
You mean by going on the camera ip and port to view the stream directly? That's an IP camera security system, you should use cameras which requires user authentication to view the streams.

jul-fls, 2022-01-07:
No the front end of motioneye doesn't ask credentials to view the cameras

jul-fls, 2022-01-07:
The cameras themselves ask credentials, no problem with that.

TyTnMonz, 2022-01-07:
I never tried it, I've seen that you can grap a single frame, but you still need to find the key ... something like http: // MotionEye: 48765 / picture / 1 / current /? _ = 1641553538092 & _username = admin & _signature = ef037ccae2a9fd3a58e432793808d4342d063cf1 shows a single frame without asking credentials, I don't know how easy it actually is to find that key. You say you can view streams without access the system? That's non good...

starbasessd, 2022-01-07:
If you set up a motionEye hub, with, say 6 network cameras, you can access each of the video streams with 
http://motioneye_ip:808x (x=camera id number)
with either admin user (able to edit all functions / settings) or user user able to view only
You shouldn't have motion web service turned on, (except for maybe localhost only) if you are concerned about 'security', and have the motion service turned on at all, as motioneye calls motion when needed. The current motioneye install instructions for motion installs then disables the service for Debian and derivatives. 
 The " http: // MotionEye: 48765 / picture / 1 / current /? _ = 1641553538092 & _username = admin & _signature = ef037ccae2a9fd3a58e432793808d4342d063cf1" 
is not motioneye, but probably HomeAssistant or something that forked motionEye. The whole point of the username and signature is to not have to log in (that's the whole username and signature purpose).
MotionEye Video Streaming Useful URL Snapshot is 
http://motioneye_ip:8765/picture/1/current/?_username=admin&_signature=\<encrypted_password\>
and requires admin login to get to it. That signature appears to be created/compared on the fly, I did a full search for mine and wasn't able to find it anywhere. If they (bad actors) can get to Settings. Video Streaming, they already have access to everything. Ditto if they can ssh in...

starbasessd, 2022-01-07:
Oops, my bad. You can access just the video stream with the  URL:808x, but that is all. The simple video stream. Not the motion triggered videos, not the snapshots. If it is a 'top secret' area, you probably shouldn't be using a free off the internet NVR system.

jul-fls, 2022-01-07:
No it's not a top secret area just corridors and tech rooms of a high school and i know but i don't have proofs that some students are hackers and pentesters so i do the maximum to protect everything IT in the school that's all.

jul-fls, 2022-01-07:
The main issue which can be resolved i think with a vlan is that this url "http://motioneye_ip:8765/" doesn't ask credentials before viewing the cameras. But to be able to configure, i need to login and whta i want is that i cannot access anything if i'm not logged i even disabled video streaming of the cameras because i will watch them with the integrated motion eye web ui and triggered movies, that's all

starbasessd, 2022-01-07:
motionEye may not be the best option for you, then. It's really designed for home / personal use, not a high school full of hackers. Another option, assuming you are using modern, managed switches on your network, would be to create a separate VLan for motionEye, and use ACLs on it.

starbasessd, 2022-01-07:
The URL "http://motioneye_ip:8765/" DOES ask for a login. the URL "http://motioneye_ip:808x/" does not.

jul-fls, 2022-01-07:
> The URL "http://motioneye_ip:8765/" DOES ask for a login. the URL "http://motioneye_ip:808x/" does not.

not in my case for the first url since the second is disabled in my case.

> motionEye may not be the best option for you, then. It's really designed for home / personal use, not a high school full of hackers. Another option, assuming you are using modern, managed switches on your network, would be to create a separate VLan for motionEye, and use ACLs on it.

Yes i use Aruba switches everywhere so i can put all the cameras and motioneye in this specific vlan, yes that's an option

STrRedWolf, 2022-01-07:
Julian, you should "weaponize" some of those students (turn them white-hat). Are you familiar with this:

https://whitehoodhacker.net/posts/2021-10-04-the-big-rick

starbasessd, 2022-01-07:
If "http://motioneye_ip:8765/" is not asking for a password, it's because you checked "Remember Me" when logging on previously. That can be cleared by clicking 'Change User' at the top right, and logging in without checking Remember Me.

jul-fls, 2022-01-07:
> Julian, you should "weaponize" some of those students (turn them white-hat). Are you familiar with this:
> 
> https://whitehoodhacker.net/posts/2021-10-04-the-big-rick

Yes i know this story we have the same type of people in this high school (in france btw) but we hopefully don't have connected monitors / projectors

jul-fls, 2022-01-07:
> If "http://motioneye_ip:8765/" is not asking for a password, it's because you checked "Remember Me" when logging on previously. That can be cleared by clicking 'Change User' at the top right, and logging in without checking Remember Me.

yes that what i would also have think but no because in private navigation it does the same

starbasessd, 2022-01-07:
No need to move cameras into the same VLAN, just ACLs VLAN maps...

jul-fls, 2022-01-07:
It seems but not sure that Aruba Switches don't have ACLs of VLANs but that's not a problem beacuse i have another option that's to put all cameras and motioneye in a subnet class C of the big class B network and tell the router to block any request from other ranges to this one and that's all so that this class C subnet is isolated from the others

starbasessd, 2022-01-07:
I have my user/password saved in FF, and open a Private window, it prompts.
Works as expected in Chrome.
Works as expected in Edge. 
Which browser?

jul-fls, 2022-01-07:
Chrome 97.0.4692.71

jul-fls, 2022-01-07:
Maybe a configuration problem

jul-fls, 2022-01-07:
It promps only when i click on the person icon not at loading the page

starbasessd, 2022-01-07:
https://community.arubanetworks.com/blogs/esupport1/2020/03/30/create-an-acl-to-allow-unidirectional-communication-between-multiple-vlans
Same version of Chrome, no issues. Private prompts for user/password every time, and doesn't affect the normal version.

jul-fls, 2022-01-07:
Ok i will be looking into that (for the vlan thing) and for the credentials i will be looking into that by reverting some configuration that i've done maybe something is wrong.

STrRedWolf, 2022-01-07:
Julien, my point is to have the students doing the pentesting follow responsible disclosure.  

Besides, if you "think" you have no such items, double check, because "think" is different from "know".  If the students find something wrong (like some administrator putting in those connected speakers w/o IT involvement), they should disclose it to you.  _70 Maxims of Maximally Effective Mercenaries number 2: A Sergent in motion outranks a Lieutenant who doesn't know what's going on._

Zixim, 2023-08-26:
+1 for multiple users, with admin selecting which user can view which footage.

MichaIng, 2024-06-11:
Another request/vote: #3006

xj25vm, 2025-08-08:
+1 for this - it seems this would also solve the problem of users affecting each other's viewing settings. At the moment if one user in a browser or phone app changes the number of columns, resolution dimmer or fps dimmer, all others connected to the same MotionEye instance are affected