motioneye-project/motioneye #542: Movie is freezing

onlize, 2017-05-26:
Hello, when the motion is detected, the movie is saved on Google Drive. The problem is that most of the movies (not all of them) are freezing for 5-15 seconds. During that time, I see the still picture and, suddenly it continues working. Basically, if someone is moving, the picture would freeze and, after 5-15 seconds, the person/animal would suddenly appear in different corner of the movie.

Thank you.

rahlzel, 2017-06-01:
I'm getting this, too. When I look at my recorded videos in VLC Media Player, they very often freeze for 5-20 seconds before suddenly fixing themselves, but the footage is now 5-20 seconds later, causing a lot of missed footage. It's like chunks of the video are only one continuous frame.

* [Canakit Raspberry Pi 3](https://www.amazon.com/CanaKit-Raspberry-Complete-Starter-Kit/dp/B01C6Q2GSY/ref=sr_1_3?ie=UTF8&qid=1496353712&sr=8-3&keywords=canakit)
* [Arducam Camera](https://www.amazon.com/gp/product/B012QRGUCQ/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1)
* motionEye Version:	0.35.2
* Motion Version:	4.0.1+git37b3595
* OS Version:	motionEyeOS 20170329

ccrisan, 2017-06-02:
I'm afraid there's nothing motionEye can do about that. There has been a lot of work in this area on the motion project repo and I believe the next release of motion will bring many fixes related to video encoding, ffmpeg and other stuff.

closed 2017-06-02 by ccrisan