motioneye-project/motioneye #544: not a supported network camera

ilmincione, 2017-05-28:
hello, I've a problem with motionEye, I'll explain the situation:

I've a dvr with several cameras attached to it, now this DVR have a "mobile" section where I can setup the access for external device. I'm using noip dns, so essentially I've configured a port forwarding to access to the DVR. The ip looks like: http://myhostname.ddns.net:18600/ 
With the android application I can setup correctly the following configuration:

hostname: myhostname.ddns.net:
port: 18600
username: admin
password: mypassword

and I can see all the camera.

In motion I've two issue, the first if I insert:

http://myhostname.ddns.net:18600/ this will return the following message:
"request timed out" (and this is really weird)

the second issue is: if I insert as url http://myhostname.ddns.net/ I get "not supported network camera".

I really don't know how can I figure out of this, someone could help me please?

ccrisan, 2017-05-28:
I seriously doubt that your DVR exposes your stream at http://myhostname.ddns.net:18600/. This is most likely the URL to the web frontend of your DVR, while the actual MJPEG URL (if any) ought to be something more complex. This is probably the reason why you get "not a supported network camera" error.

Also make sure that you can access http://myhostname.ddns.net:18600/ from your local network. Your router must support NAT reflection for you to locally access your machine via NAT (port forwarding) which is otherwise normally accessible from outside. This can be the reason why you get the timeout you mention.

ilmincione, 2017-05-28:
Hi, thanks for the reply. 

If I type in the browser this url: http://myhostname.ddns.net/ I can see the camera (only on Internet Explorer 'cause the DVR control panel use ActiveX), but yes is the frontend of the camera, else if I type 192.168.100 I can enter in the configuration of the dvr, so yes you're right about the frontend.

In the application for Android I type as port 18600 that I've forwarded in the router, and as host the ip that I've mentioned before: myhostname.ddns.net

If I type http://myhostname.ddns.net:18600/ or 192.168.1.100:18600 I'll get connection refused.
Any chance to fix this?

ccrisan, 2017-05-29:
I'm sorry but this seems like a network-related issue that has nothing to do with motionEye. I tried to help but things are a bit more complex and may require further investigation into your DVR and your network setup.

I'll close this issue as there's nothing motionEye can do to fix your problem.

closed 2017-05-29 by ccrisan