motioneye-project/motioneye #546: Brightness of on RPI Zero W with PiCam NoIR 2?

cyclops1982, 2017-05-31:
Hi,

i've been doing some testing and so far, i'm very impressed with MotionEye!

On the raspbery pi zero, i can see that high CPU load comes very quickly with the size of the iamge and the framerate. I'm ok with low framerate! I also noticed that 'save movie only' works quite well too.

My problem is that if i change the resolution higher than 800x600 that the image brightness and other settings are completely off. Ideally, i'd like an 1080P image, but at the moment only 800x600 seems to give the correct colours and brightness. I've tried auto brightness as well as some other settings.

What could cause this? I took some photo's with the normal pi commands and they all seem ok.

Ringostar1982, 2017-07-07:
maybe the same Problem like mine? 

[https://github.com/ccrisan/motioneye/issues/575](url)

JamBalaya56562, 2026-11-07:
Closing, as this went unanswered here; the picture going wrong above a certain resolution with the camera module is also discussed in #575. On current Raspberry Pi OS the camera module goes through libcamera, see [Troubleshooting](https://github.com/motioneye-project/motioneye/wiki/Troubleshooting#libcamera-camera-stack). Please open a new issue if the picture is still off at higher resolutions with a current release.

closed 2026-11-07 by JamBalaya56562