motioneye-project/motioneye #547: MotionEyeOs and Reverse Proxy Nginx

Lakemast, 2017-06-01:
Hello crisan,

First of all, I want to thank you for this beautiful work!
I'm trying to use MotionEyeOs behind a Reverse Proxy Nginx
I followed this tutorial [Running Behind Nginx](https://github.com/ccrisan/motioneye/wiki/Running-Behind-Nginx]), but I'm getting this:

[image]

Here is my Nginx conf file:
```
server {
       listen         80;
       server_name    arvim.ddns.net;
       return         301 https://$server_name$request_uri;
}
server {
    	listen      443 ssl;
    	server_name arvim.ddns.net;
    	ssl_certificate     /home/pi/servidor_arvim/ssl/nginx.crt;
    	ssl_certificate_key /home/pi/servidor_arvim/ssl/nginx.key;
    	ssl_protocols       TLSv1 TLSv1.1 TLSv1.2;
    	ssl_ciphers         HIGH:!aNULL:!MD5;
    	charset     utf-8;
    	client_max_body_size 75M;
	
    	location / { try_files $uri @yourapplication; }
    	location @yourapplication {
        				include uwsgi_params;
        				uwsgi_pass unix:/home/pi/servidor_arvim/arvim_uwsgi.sock;
    	}
	location /cameras/ {

			proxy_pass http://192.168.1.107/;
			proxy_read_timeout 120s;
        		access_log off;

	}
}
```

What am I doing  wrong?

Thanks in advance!

Lakemast, 2017-06-01:
I don't know what i have done, but now it's working!
I guess adding "/cameras/" to BasePath fixed it!
Thanks a lot crisan! You have done a great job!

closed 2017-06-01 by Lakemast

cslev, 2021-09-21:
Hi, I am still facing this issue!

I have an nginx reverse-proxy conf like this:
```
...
set $meye http://172.30.1.5:8765/;
    location /meye/ {
        proxy_pass $meye;

#        proxy_set_header Host $http_host;
#        proxy_set_header Host $host;
#        proxy_set_header Upgrade $http_upgrade;
#        proxy_set_header Connection "upgrade";

#        proxy_set_header X-Real-IP $remote_addr;
#        proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
#        proxy_set_header X-Forwarded-Proto $scheme;
#        proxy_set_header X-Forwarded-Host $host;
#        proxy_set_header X-Forwarded-Port $server_port;
#        proxy_http_version 1.1;
#        proxy_buffering off;
#        client_max_body_size 0;
        proxy_read_timeout 120s;
#        proxy_redirect $meye /;    
    }
...
```
Yeah, I tried a lot of configs, just let them there with comments.
The IP I have for motioneye is a docker IP that can be accessed directly within the system.

I use nginx proxy to access my camera (outside my home network via port forwarding to nginx) with the <DOMAIN_NAME>/meye (or <DOMAIN_NAME>/meye/), whichever works.

I followed the instructions here and through the links, but still having the issue of not loading js/css and other resources.

I have found that there is a `server_name` setting in the `motioneye.conf` in my container under `/etc/motioneye/motioneye.conf`. 
That actually says that I have to use it and configure it if I want reverse proxy. So, i tried with <DOMAIN_NAME> and also with <DOMAIN_NAME>/meye.

What else should I try?
thanks

starbasessd, 2021-09-21:
@cslev You may want to open a new issue, as this one was last commented on and closed over 4 years ago. Be sure to include your current info.