motioneye-project/motioneye #548: Feature request: two-factor authentication

stfro, 2017-06-01:
Hello, 

I love MotionEye for 1000 reasons, but I do dislike the fact that there isn't any two-factor authentication available. Having the camera exposed towards the Internet, I think it's just a matter of time until somebody hacks in. 
Or if anybody haves a better idea for making sure that nobody else except for me can access the camera, please share your ideas. 

Stefan 

PS: also an app to determine my location using geo fencing and disable the camera when I'm home would be nice.

ccrisan, 2017-06-02:
I'm sorry but this is not something that's going to happen anytime soon. We have other priorities such as improving the performance, porting to other boards and whatnot. If someone sends in a PR with such a feature, I'd be happy to review it though, but I'm afraid it's not going to be something easy to implement.

closed 2017-06-02 by ccrisan

Bradley-Parker, 2017-06-14:
@stfro If you are concerned about security you will want to limit _any_ internet exposed applications you run on your network. There are plenty of articles online about setting up VPNs or SSH tunnels. Using one of those methods you only have one exposed service that is built solely for secure communications.

surak, 2020-06-04:
TOTP has gone a long way since then. This is something I would rethink in 2020.