motioneye-project/motioneye #550: Suggestion for "Delete"

DonGrego617, 2017-06-07:
Hi, 
Great app, love it!!  Just one suggestion?  In the "play video" section, where I can click on a video and "preview" it...and the preview enlarges and show's a really large image of the video - you have the two options:  [CLOSE] and [DOWNLOAD].  

It would be SUPER USEFUL if there was a third button there for [DELETE].  I find myself having to go back and forth between the enlarged picture to preview it to see if it's worth keeping or deleting, but then having to close that preview window, remember which video I just previewed, then clicking "DELETE", then clicking the next video to enlarge and preview the next one, close it, delete it, open next, preview it, close it, delete it.  lol.

You get the point  Just a usability suggestion!  Thanks!!
Don

zoidy, 2019-05-20:
This would be great!

CRImier, 2019-05-31:
There's a button to upvote the original comment instead of adding yet another "+1" comment

[image]

PR #1833 (2020-07-08): Add Delete button to picture dialog
This closes #550, closes #1309, closes #1409 and closes #1425.
closed 2020-07-08 by ccrisan