motioneye-project/motioneye #554: motion rapstill

pierrotori, 2017-06-11:
Hi

if i undestand when the motion service is running i can't to launch the command raspistill
is it possible to push a command to create a photo with the daemon service?

pierrotori

chrisf4lc0n, 2017-06-12:
https://github.com/ccrisan/motioneye/issues/94

pierrotori, 2017-06-12:
thanks a lot
i try the command curl http://XXX.XXX.X.X:8080/0/action/snapshot and it's ok now
[http://www.lavrsen.dk/foswiki/bin/view/Motion/MotionHttpAPI](url)
the snapshot name is the name in the motion.conf, it's perfect

Now i can compare a reference snapshot with the snapshot at the time
my scenario for the water in my garden i check two snapshots if the difference  is nul i can start the water

JamBalaya56562, 2026-11-07:
Closing as solved, as you confirmed: the `action/snapshot` URL of Motion's HTTP API takes a picture while the daemon is running, without `raspistill`.

closed 2026-11-07 by JamBalaya56562