motioneye-project/motioneye #556: embed url not working after ugrade to 0.36

myB0B, 2017-06-12:
trying to connect to http://xxx.xxx.xxx.xxx:8765/picture/1/frame/
result of 19 errors 404 in chrome log.

chrisf4lc0n, 2017-06-15:
Have you tried to clear the cache?

myB0B, 2017-06-15:
yes of course
test with chrome 59.0.3071.86, edge 38.14393.1066.0 , firefox 53.0.3
idem with chrome for android

GET http://myip:8765/picture/1/frame/static/css/jquery.timepicker.css 
myip/:21 GET http://myip:8765/picture/1/frame/static/css/ui.css 
myip/:30 GET http://myip:8765/picture/1/frame/static/js/jquery.min.js 
myip/:29 GET http://myip:8765/picture/1/frame/static/js/css-browser-selector.js 
myip/:24 GET http://myip:8765/picture/1/frame/static/css/frame.css 
myip/:22 GET http://myip:8765/picture/1/frame/static/css/main.css 
myip/:31 GET http://myip:8765/picture/1/frame/static/js/jquery.timepicker.min.js 
myip/:32 GET http://myip:8765/picture/1/frame/static/js/jquery.mousewheel.js 
myip/:38 GET http://myip:8765/picture/1/frame/static/js/ui.js 
myip/:39 GET http://myip:8765/picture/1/frame/static/js/main.js 
myip/:41 GET http://myip:8765/picture/1/frame/static/js/frame.js 
myip/:62 GET http://myip:8765/picture/1/frame/static/img/no-camera.svg 404 (Not Found)
(index):30 GET http://myip:8765/picture/1/frame/static/js/jquery.min.js 
(index):31 GET http://myip:8765/picture/1/frame/static/js/jquery.timepicker.min.js 
(index):32 GET http://myip:8765/picture/1/frame/static/js/jquery.mousewheel.js 
(index):38 GET http://myip:8765/picture/1/frame/static/js/ui.js 
(index):39 GET http://myip:8765/picture/1/frame/static/js/main.js 
(index):41 GET http://myip:8765/picture/1/frame/static/js/frame.js 404 (Not Found)
no-camera.svg:1 GET http://myip:8765/picture/1/frame/static/img/no-camera.svg 404 (Not Found)

ccrisan, 2017-06-16:
There's appears to be a regression bug with the relative paths change and the frame URL. I'll make sure to fix it asap.

myB0B, 2017-06-17:
by the way, cloud you add CORS in tornado http headers ?

ccrisan, 2017-06-17:
https://github.com/ccrisan/motioneye/commit/500ed86de1eadefaeb4fe81d90440edced980a92 fixes this issue. I'm not sure about CORS but I accept pull requests :)

closed 2017-06-17 by ccrisan