motioneye-project/motioneye #558: Video flickers in latest WebKit-based development browsers

legoadk, 2017-06-14:
Hello,

I'm not entirely sure if this is a WebKit issue or something that can be mitigated by motionEye, but here goes.

I'm running motionEyeOS 20170611, and I have Google Chrome Canary 61.0.3130.0 installed on my Mac, and iOS 11 Developer Preview (with Safari 602.1/AppleWebKit 604.1.21) installed on my iPhone.

When viewed in _either_ of these browsers, the video feed "flickers" from transparent to opaque each time a frame is loaded from the server, revealing whatever is behind the image - whether it be the red "motion detected" background, or the normal grey background. This appears as a very rapid flicker, and can vary in speed based on the FPS achieved by the server/camera.

This issue does _not_ occur on older versions of Chrome Canary and iOS (though I can't pinpoint when, exactly, it became an issue).

As these browsers represent the future of their "stable" releases, I figure it may be wise to let you know of the issue now -- before it becomes an issue when these browsers "hit store shelves" (if they don't fix the issue in the browser side).

ccrisan, 2017-06-15:
I'll look into it. Thanks for the heads up.

JamBalaya56562, 2026-11-07:
Closing, as this was seen in developer previews of Chrome 61 and iOS 11, and there were no further reports here once those were released. If the live picture still flickers with a current browser and motionEye, please open a new issue with the browser version.

closed 2026-11-07 by JamBalaya56562