motioneye-project/motioneye #560: Motioneyeos: "an error occured refreshing recommended" while trying to add camera

fcooks, 2017-06-15:
I have a raspi running motioneye as hte frontend and a few raspi boards with camera modules around the house. I had all cameras discovered no issues. I deleted one camera that was running motion and motioneye which i installed from scratch . I decided to use motioneyeos on this cam . 
After imaging the cam with Motioneyeos i can no longer discover it as a remote camera using the frontend.

This is strange because i use the same image on a new card that I have used for all my other cams so the image is working fine. all other cams were discoverd. 
I have tried restarting the frontend host and the services on the cam itself but no resolution ...

Open to any ideas here :-)

ccrisan, 2017-06-15:
Please attach your motioneye.log file.

fcooks, 2017-06-18:
HI Calin 

attached find the motioneye.log file from the camera itself. 

I have run a tail -f on the log file of the camera and the raspi running the motioneye frontend and found something very interesting. 

On the host running the motioneye frontend when i add a camera and view the log file it does the following. 
1) even though teh camera address has changed to 192.168.1.101 where in the past it was 192.168.1.100, the log file seems to get a request from the raspi discovering it with the following  http://192.168.1.100:8765.
this is not the address i enter to discover the cam. ....so it seems the raspi running hte motioneye frontend keeps trying to discover the cameras old address on x.x.x.101:8765 even though i deleted that cam using the gui. so somehow that address seems to stick in the cache on the raspi hosting the frontend. as this is the request turning up on the camera all the time......not sure where to clear that ?

Cheers
Fritz

2017-06-05 03:29:12: [motioneye]     INFO: hello! this is motionEye server 0.35.2
2017-06-05 03:29:13: [motioneye]     INFO: main config file /data/etc/motion.conf does not exist, using default values
2017-06-05 03:29:13: [motioneye]     INFO: cleanup started
2017-06-05 03:29:13: [motioneye]     INFO: wsswitch started
2017-06-05 03:29:14: [motioneye]     INFO: tasks started
2017-06-05 03:29:14: [motioneye]     INFO: mjpg client garbage collector started
2017-06-05 03:29:14: [motioneye]     INFO: smb mounts started
2017-06-05 03:29:14: [motioneye]     INFO: server started
2017-06-05 03:29:15: [motioneye]     INFO: adding new camera with id 1...
2017-06-05 03:29:16: [motioneye]     INFO: main config file /data/etc/motion.conf does not exist, using default values
2017-06-05 03:31:06: [motioneye]     INFO: hello! this is motionEye server 0.35.2
2017-06-05 03:31:10: [motioneye]     INFO: cleanup started
2017-06-05 03:31:10: [motioneye]     INFO: wsswitch started
2017-06-05 03:31:10: [motioneye]     INFO: tasks started
2017-06-05 03:31:10: [motioneye]     INFO: mjpg client garbage collector started
2017-06-05 03:31:10: [motioneye]     INFO: smb mounts started
2017-06-05 03:31:10: [motioneye]     INFO: server started
2017-06-05 03:33:09: [motioneye]     INFO: hello! this is motionEye server 0.35.2
2017-06-05 03:33:13: [motioneye]     INFO: cleanup started
2017-06-05 03:33:13: [motioneye]     INFO: wsswitch started
2017-06-05 03:33:13: [motioneye]     INFO: tasks started
2017-06-05 03:33:13: [motioneye]     INFO: mjpg client garbage collector started
2017-06-05 03:33:13: [motioneye]     INFO: smb mounts started
2017-06-05 03:33:14: [motioneye]     INFO: server started
                                                                                                                                                                                                                                                                                                                                                                                  2017-06-05 03:49:00: [motioneye]     INFO: hello! this is motionEye server 0.35.2
2017-06-05 03:49:04: [motioneye]     INFO: cleanup started
2017-06-05 03:49:04: [motioneye]     INFO: wsswitch started
2017-06-05 03:49:04: [motioneye]     INFO: tasks started
2017-06-05 03:49:04: [motioneye]     INFO: mjpg client garbage collector started
2017-06-05 03:49:04: [motioneye]     INFO: smb mounts started
2017-06-05 03:49:05: [motioneye]     INFO: server started
2017-06-05 06:02:51: [motioneye]  WARNING: 403 GET /login/?_=1496642572245&_username=_&_signature=3eb9d1373dbf311adb81a872723a33bccc97611c (192.168.1.68) 19.08ms
2017-06-05 06:03:16: [motioneye]  WARNING: 403 GET /login/?_=1496642597535&_username=_&_signature=6971d057492b6eb08522df0fa1bfcc0ac855393a (192.168.1.68) 12.68ms
2017-06-05 06:03:49: [motioneye]  WARNING: 403 POST /config/0/set/?_=1496642630438&_username=admin&_signature=263d058a807f6b24c7c7b87e5d85deb56ec1dfac (192.168.1.68) 22.61ms
2017-06-05 06:03:49: [motioneye]  WARNING: 403 POST /config/0/set/?_=1496642630384&_username=admin&_signature=30a29807da698b938fc5cff142850f5fcce8806d (192.168.1.68) 20.42ms
2017-06-05 06:03:50: [motioneye]  WARNING: 403 GET /picture/1/current/?_=1496642630724&_username=admin&_signature=65f1a176a2c16c4e2b221f8ff36bb5bd299c80c5 (192.168.1.68) 16.01ms
2017-06-05 06:03:51: [motioneye]  WARNING: 403 GET /picture/1/current/?_=1496642631933&_username=admin&_signature=7b3c85a85b223b697ed0a081bdd82c59abd2e964 (192.168.1.68) 16.47ms
2017-06-05 06:03:51: [motioneye]     INFO: rebooting
2017-06-05 06:03:52: [motioneye]     INFO: executing "/sbin/reboot"
2017-06-05 06:03:52: [motioneye]  WARNING: 403 GET /picture/1/current/?_=1496642633056&_username=admin&_signature=05522fb16fe6e8648f353a75c420013cb88e4ab9 (192.168.1.68) 56.17ms
2017-06-05 06:03:53: [motioneye]  WARNING: 403 GET /picture/1/current/?_=1496642634140&_username=admin&_signature=04f3c1652e93e9adcc84a4bebc9daab1fc10c31e (192.168.1.68) 26.15ms
2017-06-05 06:03:54: [motioneye]  WARNING: 403 GET /picture/1/current/?_=1496642635210&_username=admin&_signature=368bbdc49c6b7aefa17810b637081651fe6720c0 (192.168.1.68) 28.93ms
2017-06-05 06:03:56: [motioneye]  WARNING: 403 GET /picture/1/current/?_=1496642636324&_username=admin&_signature=0b2d006bb1afb8427085376e17e1735f843b5c79 (192.168.1.68) 50.66ms
2017-06-05 06:03:56: [motioneye]     INFO: interrupt signal received, shutting down...
2017-06-05 06:03:56: [motioneye]     INFO: server stopped
[... cut ...]
2017-06-18 15:18:59: [motioneye]    ERROR: failed to get config for remote camera 1 on Http://192.168.1.100:8765/config/1: Failed to connect to 192.168.1.100 port 8765: No route to host
2017-06-18 15:19:23: [motioneye]  WARNING: 403 GET /config/list/?_username=admin&_admin=true&_signature=fe1c3bb548cf03550e88f0c38843a00c9f7b2d8a (192.168.1.68) 16.27ms
2017-06-18 15:19:34: [motioneye]    ERROR: failed to get config for remote camera 1 on Http://192.168.1.100:8765/config/1: Failed to connect to 192.168.1.100 port 8765: No route to host

ccrisan, 2017-06-19:
I'm sorry but this is overly complicated. It also seems like a network configuration issue at your side. I can't help you with that. Anyways, the next time you attach a log file, please *attach* it, don't paste it - it's very difficult and unpleasant to read through such pasted contents.

closed 2017-06-19 by ccrisan