motioneye-project/motioneye #563: Trigger recording from outside Motion(eye)

antares2001, 2017-06-16:
Hello @ccrisan 

after a long time I've come back to give a try to Motion Eye and I like it a lot, when running on my Debian machine. Yet, since I am using more than a hand full of cameras, I cannot use software based motion detection as it is a) far too unreliable and b) it eats up all my CPU already with 2-3 cameras.

Hence, I've recently been playing with using my home automation motion sensors to trigger recording, by calling a URL. I've tried it with Zoneminder and it worked quite well, as they offer http URLs that can be used to start and stop recording (web-triggered recording they call it). Yet, Zoneminder was far too fat and the lack of recording movies instead of .jpgs makes it no feasible solution for me (hard disk runs full after a day of recording and the filesystem goes crazy).

I'd like to use Motioneye for that purpose, but I've not been able neither by using Google nor Github search to find how to trigger recording externally.

Is there any API to motioneye that allows it? Can I trigger and stop recording from a camera by calling an API? I've seen that I can see the stream of camera 1 via http://server:8081 - so a http://server:8081?recording=[start|stop] would be a great way to control recording.

Yet, I am afraid this is not the case, and hence I'd then add this as feature request which would make Motioneye really a superior system, that then can be integrated with more reliable motion sensors than software based motion-detection is (and this is also much more energy efficient and saves the environment :-)).

frekel, 2017-06-29:
I second this request!

Fmstrat, 2017-07-09:
I third. This is the one reason I can't use motioneye. Could really use a way to trigger a recording. Start/stop might be problematic in case a call fails, especially with sensors that don't report an "off" event for motion. Instead, I would suggest:

```
http://server:8765/capture?camara=NAME&duration=DURATION
```

Where `NAME` is the name of the camera and `DURATION` is the number of seconds to record.

Fmstrat, 2017-07-09:
After some more investigation, this may not be possible due to the limitations of how the daemon `motion` works. Oh well.

peterwtux, 2017-07-12:
Can't you use motion's webcontrol HTTP interface to trigger the recording? https://github.com/ccrisan/motioneye/issues/31#issuecomment-314634755

dtbaker, 2018-11-05:
Hey I set this up yesterday FYI

Start motion recording:
```curl "http://localhost:7999/1/config/set?emulate_motion=1```
Disable motion recording:
```curl "http://localhost:7999/1/config/set?emulate_motion=0```

maaikez, 2018-12-09:
@dtbaker That's great! We just installed the docker container. Motion and motion eye are great! Much better than zoneminder. We tried shinobi as well, but not happy with it.

However, what I wanted to say is that it starting motion recording with this get request seems to work perfectly. Thank you!

MYeager1967, 2020-05-19:
Does anyone know if starting a recording this way triggers the motion notification section of motionEye???

Winston70, 2021-04-24:
This also puzzles me. It's brilliant piece of software. I read that everywhere. However, once installed, I had a question. How do you actually make it work?

starbasessd, 2021-04-24:
What do you mean, 'make it work'?
What do you want it to do?
1st, you install it to SDCard (or if using dev20201026 a USB Thumb Drive)
2nd, you copy the ssh or ssh.txt file and wpa_supplicant.conf file to the /boot partition
3rd, you insert the SDCard or Thumb drive into a Pi (or similar supported SBC)
4th, you connect to the WebGUI at http://[ip_address]
5th, you add IP Cameras, or USB Web Cameras, or CSI PiCams to the system
6th, you set up where you want the files saved (locally, NAS, GDrive or Dropbox among options)
7th, you set up how sensitive you want motion detection, whether to do snap shots, videos, what format videos, etc.
8th, enjoy.
The above curl commands talk directly to motion, the back end recording program. motionEye is the front end graphical user interface, to make most options easier to set.
In the other ticket/issue, I mentioned you could use the Action Buttons to call scripts. you could create those scripts and use those commands in them...
Without knowing anything about programing you could use motion/motionEye or motionEyeOS and never even look at the underlying system, like a plug'n'play home NVR security system.

Winston70, 2021-04-24:
By make it work I mean start and stop recording. I don't know what a curl
command is. I'm not certain what a script is, let alone write or 'call'
one. This is one of least user friendly apps I come across. And for linux
that is saying something. I admit that I am a windows retard, but I can't
understand why no one can understand what I'm asking. I can't add external
storage because I have no idea how to access it. If there is a point I'm
missing, please tell me. Is this app a joke on non linux hardcores? Is this
a closed club that non linux users can't join? After I setup cameras and
storage, does it just start recording?

starbasessd, 2021-04-24:
I'm sorry, but motionEye/motionEyeOS are probably not for you, then. It's
not a 'club'. It is open source, so, if a process exists you can learn from
it. If a process doesn't exist, you can learn about it and add your own
processes.
You may want to start off at one of the teaching/learning sites like
http://raspberrypi.org, or http://learn.adafruit.com, where they take you
step by step through basic Linux, Python, scripting, and how to use
Raspberry Pi hardware. It might also be a good thing to not bandy about
phrases like "Is this app a joke on non linux hardcores? Is this a closed
club that non linux users can't join?" If you ask specific questions, I or
another user can probably help you. Ask a generic question like 'how do I
make it work', sounds like you want front to back usage instructions. For a
non-Mac person, or a non-Windows person, or even yes, a non-linux person,
it indicates you don't want to learn by doing, in which case I could point
you to a number of pre-built security camera companies with very nice
cameras and NVRs, but then you wouldn't be able to take it apart, look
under the hood, or modify it to suit your needs.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Winston70, 2021-04-25:
it's tough to learn by doing when you have no idea what you are supposed to
do. This app has the potential for very wide spread use but without better
end user instructions, it's doomed to the pile of old linux projects that
people get tired of trying to make work. Why should I feel like I need to
go to community college just to make an app function? I don't mind learning
by doing. I had a laptop running Mint and it took a long time, but it
worked  properly. I spent hours just trying to get it to find a usb wifi
adapter. But I did it and it worked fine. I found out that pi zero w is
shit. Most of my post came from frustration. I'm afraid I don't cope well
with things that don't work. Ok, it works, I just don't have the mindset to
learn how it operates. Creators need to understand that there are end users
that want to use their apps, but are not inclined to dig into the guts.
Call me a windows spoiled user, but give credit to MS, their stuff works. I
know it's absurd to get enraged over some little chicken shit linux app.
But would it kill creators to make their apps a little more user friendly?
I'm 66 years old and would like things that just work. Does that make me
unreasonable? If so, then I'm unreasonable. What is so wrong with an on
screen indicator that it is recording? How about a button on screen to say
'start' or 'stop'?

MYeager1967, 2021-04-25:
Sounds to me like you don't understand a few basic things. First, MotionEye
is a front end for Motion. Motion is an app from a completely different
developer and is very well documented. MotionEye is also well documented.
Don't blame the developer of one app for your inability to understand a
different app....

starbasessd, 2021-04-25:
I, also, am 60 y/o, and support motioneye & motionEyeOS for free as a volunteer, and beta test other apps as requested.
I have a nice selection of hardware & software to test other's setups (like yourself) to reproduce issues for the programmers and developers.
The indicator that motion has triggered recording is a red outline box around the video, and/or if you turned on Show Frame Changes in Settings, Motion Detection
I am willing to sit here day after day walking users with issues through learning how to use products (witness the longer threads over the past 18 months here and in the motionEyeOS github). I make/made my living as an Enterprise Third Level Help Desk & Systems Administrator for M$ and Linux. If my end user is willing to learn, I will sit on the other end of the log all day long. M$ is one company, who controls their environment, and will tell you in a heartbeat to go to the 3rd party product support. Linux has a few companies, but nothing like Apple & M$, thousands of flavors of desktops and servers, hundreds of thousands of applications, and millions of users that love the freedom to pick and choose.
Be that as it may, with the hostility you are showing, I will close and lock this conversation.

closed 2021-04-25 by starbasessd