motioneye-project/motioneye #564: Not able to add Vstarcam web-cam with rtsp, but rtsp works with VLC.

TheFryingPI, 2017-06-17:
Hi,
I am trying to add my Vstarcam webcams to MotionEye with rtsp, the rtsp works fine with VLC, but in the VLC URL is the login details embedded, not sure if this is the issue, but I hope someone can help me and explain why it is not working.

I add a new camera, select Network Camera, url: rtsp://192.168.0.251:554/tcp/av0_1, enter admin, password, select RTSP/TCP camera, enter, it saves camera and tries to connect, but screen shows Unable to open device. The VLC URL is having user admin included, perhaps there is the problem, but I am not sure in how to debug or find any log in motioneye to get some clarity.

VLC URL: rtsp://admin:123456@192.168.0.251:554/tcp/av0_0
[image]

[image]
[image]

Vstarcam information:
Main stream:              rtsp://IP:554/udp/av0_0       1280 x 720
Secondary stream:         rtsp://IP:554/udp/av0_1        640 x 480
The third stream:         rtsp://IP:554/udp/av0_2        320 x 240
Secondary stream + Audio: rtsp://IP:554/udp/av1_1  	 640 x 480

Note: All these four streams with either UDP or TCP are working perfect in VLC.
Although the camera gives possibility for other users only the admin is able to stream rtsp.

Camera settings:
[image]

This is how it looks in the Firefox browser:
[image]

Any help is much appreciated, as I would like to have it all working with my Raspberry Pies.

chrisf4lc0n, 2017-06-18:
Have you tried the VLC URL rtsp://admin:123456@192.168.0.251:554/tcp/av0_0 without credentials in MotionEye?

TheFryingPI, 2017-06-19:
@chrisf4lc0n 
I tried that way as well, but it won't allow me to add the new camera including the admin:12356@ in the camera url.

TheFryingPI, 2017-06-19:
@chrisf4lc0n 
I have also tried to modify the camera settings, but that is not possible in the gui, so I need to delete the camera and add new one and try each time with different settings, is it perhaps a file-right issue not being able to modify the  camera? It seems odd to me because if I for whatever reason change the password for that camera I need to delete it and add new :-(

TheFryingPI, 2017-06-19:
@chrisf4lc0n 
I tried again and again, now this time I got error: Protocol "rtsp" not supported or disabled in curl.

chrisf4lc0n, 2017-06-20:
The ONVIF stream http url... Have you teied that one yet?

TheFryingPI, 2017-06-21:
@chrisf4lc0n 
Hiya, yes I tried that but it doesn't allow me to add the camera, with: Empty reply from server.
As I wanted to be sure it works with rtsp, I installed a Windows VM in Virtualbox, installed Onvif device manager and it streams perfect with rtsp.

chrisf4lc0n, 2017-06-21:
Can change the log level for MotionEye to debug and post the content while trying to add a camera, please?

TheFryingPI, 2017-06-21:
@chrisf4lc0n 
Yes of course, I will find out how to do that, in mean time I installed the BlueIris in the same windows VM and that works as well, it's very frustrating as I said goodbye to Windows and using Debian and Raspberry PI...

[image]

TheFryingPI, 2017-06-21:
@chrisf4lc0n 

Content of /var/log/messages
I removed camera and added again.

Jun 21 09:35:40 vm motion: [1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
Jun 21 09:35:44 vm motion: [1] [NTC] [ALL] motion_loop: End of event 1
Jun 21 09:35:44 vm motion: [1] [NTC] [ALL] motion_loop: Thread exiting
Jun 21 09:35:44 vm motion: [1] [NTC] [STR] stream_stop: Closing motion-stream listen socket & active motion-stream sockets
Jun 21 09:35:44 vm motion: [1] [NTC] [STR] stream_stop: Closed motion-stream listen socket & active motion-stream sockets
Jun 21 09:35:44 vm motion: [0] [NTC] [ALL] main: Threads finished
Jun 21 09:35:45 vm motion: [0] [NTC] [STR] httpd_run: motion-httpd - Finishing
Jun 21 09:35:45 vm motion: [0] [NTC] [STR] httpd_run: motion-httpd Closing
Jun 21 09:35:45 vm motion: [0] [NTC] [STR] motion_web_control: motion-httpd thread exit
Jun 21 09:35:45 vm motion: [0] [NTC] [ALL] main: Motion terminating
Jun 21 09:36:59 vm motion: [0] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Jun 21 09:36:59 vm motion: [0] [NTC] [ALL] config_thread: Processing config file thread-1.conf
Jun 21 09:36:59 vm motion: [0] [NTC] [ALL] motion_startup: Motion 3.2.12+git20140228 Started
Jun 21 09:36:59 vm motion: [0] [NTC] [ALL] motion_startup: Logging to syslog
Jun 21 09:36:59 vm motion: [0] [NTC] [ALL] motion_startup: Using default log type (ALL)
Jun 21 09:36:59 vm motion: [0] [NTC] [ALL] motion_startup: Using log type (ALL) log level (ALL)
Jun 21 09:36:59 vm motion: [0] [NTC] [ENC] ffmpeg_init: ffmpeg LIBAVCODEC_BUILD 3670272 LIBAVFORMAT_BUILD 3670272
Jun 21 09:36:59 vm motion: [0] [NTC] [ALL] main: Thread 1 is from thread-1.conf
Jun 21 09:36:59 vm motion: [0] [NTC] [ALL] main: Thread 1 is device: rtsp://192.168.0.251:554/tcp/av0_0 input -1
Jun 21 09:36:59 vm motion: [0] [NTC] [ALL] main: Stream port 8081
Jun 21 09:36:59 vm motion: [0] [NTC] [ALL] main: Waiting for threads to finish, pid: 3233
Jun 21 09:36:59 vm motion: [1] [NTC] [ALL] motion_init: Thread 1 started , motion detection Enabled
Jun 21 09:36:59 vm motion: [1] [WRN] [ALL] motion_init: Could not fetch initial image from camera Motion continues using width and height from config file(s)
Jun 21 09:36:59 vm motion: [1] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 1 items
Jun 21 09:36:59 vm motion: [1] [NTC] [STR] http_bindsock: motion-stream testing : IPV4 addr: 0.0.0.0 port: 8081
Jun 21 09:36:59 vm motion: [1] [NTC] [STR] http_bindsock: motion-stream Bound : IPV4 addr: 0.0.0.0 port: 8081
Jun 21 09:36:59 vm motion: [1] [NTC] [ALL] motion_init: Started motion-stream server in port 8081 auth Disabled
Jun 21 09:36:59 vm motion: [1] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 21 items
Jun 21 09:36:59 vm motion: [0] [NTC] [STR] httpd_run: motion-httpd testing : IPV4 addr: 127.0.0.1 port: 7999
Jun 21 09:36:59 vm motion: [0] [NTC] [STR] httpd_run: motion-httpd Bound : IPV4 addr: 127.0.0.1 port: 7999
Jun 21 09:36:59 vm motion: [0] [NTC] [STR] httpd_run: motion-httpd/3.2.12+git20140228 running, accepting connections
Jun 21 09:36:59 vm motion: [0] [NTC] [STR] httpd_run: motion-httpd: waiting for data on 127.0.0.1 port TCP 7999
Jun 21 09:37:00 vm motion: [1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera

TheFryingPI, 2017-06-21:
@chrisf4lc0n  I went through all steps again, and installed the latest version of motion. I have no more the error of not logged in, and I can see images stored and videos, so it works partly, but my screen is still gray, nothing to display, is there some display software missing?

closed 2017-07-01 by TheFryingPI

therman89, 2020-04-29:
Hi,
Did you manage to solve this issue? I have the same problem and can't get around it. I have 2 Vstarcams, which are different models. One works, the other does not. Very annoying.

Tibi

UmerFraz, 2020-05-15:
hi, can you please post our working settings ?

therman89, 2020-05-17:
This address works for me, when I add the camera:
rtsp://192.168.1.14:10554/tcp/av0_0

For me, the problem was that I changed the settings in the camera web UI. I modified the RTSP port to 554, which did not work. I reset it back to 10554 and now it's working.

fratellino69, 2020-10-04:
Same problem for me....anyone found solution???