motioneye-project/motioneye #570: cant play sound on motion detect

jdowdle, 2017-06-25:
hello, all seems good, on ubuntu 17.04, but cant get a sound on motion.
I have verified the following plays from command line:
aplay /usr/share/sounds/alsa/Front_Center.wav
I see the following reports in motion.log
[1:ml1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.11.59/defaultPrimary?streamType=u)
[1:nc1] [ALR] [NET] netcam_handler_loop: Camera handler thread [5] started
ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
aplay: main:788: audio open error: No such file or directory

I have tried numerous other sound file players with no luck.

I am very new to ubuntu, primarily a windows user.
any help would be appreciated.
thanks.

Kywalh, 2017-07-27:
Hi,
I'm not an expert... I would say even closer to a newbie but since I've tried to make it work on my RPi3, I had a mistake of the file path... 
after a quick check and a correction, it works now !!!

Are you sure the path is OK ?

electron1979, 2018-12-10:
https://github.com/ccrisan/motioneye/issues/435

closed 2018-12-17 by ccrisan