motioneye-project/motioneye #573: port number

bmili, 2017-06-29:
I've installed Motioneye OS (pi_jessie_motion_4.0.1-1_armhf.deb). When I open in browser http://192.168.1.81:8081/ there is only image from camera 1 displayed.

ccrisan, 2017-06-30:
Use port 80.

closed 2017-06-30 by ccrisan

bmili, 2017-06-30:
my Motioneye is not running on port 80 but on port 8765 (
http://192.168.1.81:8765/ ) I accidentally   found it out. I wonder where
it is from?  I didn't set up any port.

Thank you!

Regards

Boris

ccrisan, 2017-06-30:
Then you didn't install motionEyeOS, but plain motionEye on Raspbian or such.

bmili, 2017-06-30:
Months ago I installed MotioneyeOs, but kow I installed plain motionEye on
Raspbian.

I've added push notification apps to Motioneye. How can I disable
notification of moving Branches in the wind?

Thank you.

Boris