motioneye-project/motioneye #574: Can't execute action button script

unto1, 2017-07-02:
I try to show action button on my camera frame. But i cant execute the script. I try to insert text to my text file
I manually execute the script on my terminal its success. But when i click action button i camt success to execute the script. 
Here the detail video.
https://youtu.be/9gJ2UuGNAuQ

ccrisan, 2017-07-04:
I need to see your motioneye log file (usually obtainable by running `journalctl -u motioneye > motioneye.log`.

unto1, 2017-07-05:
hi crisan, 
thanks for your fast respons. 
here the attach motioneye.log file

[motioneye.log.txt](https://github.com/ccrisan/motioneye/files/1123457/motioneye.log.txt)

ccrisan, 2017-07-05:
When you're using shell scripts, you must start the file with a *shebang*:

    #!/bin/bash

closed 2017-07-05 by ccrisan