motioneye-project/motioneye #575: green tinted images

Ringostar1982, 2017-07-05:
Recently i noticed that my Raspi Cam Images are **green tinted**.

If i change Brightness to 100% in GUI an back to default(50%) it looks all good again, 
but after a few hours/days the image/video is **green tinted again** !

Updating to the newest versions did not help.
Video Resolution: 1280x1024

Any ideas?
Thanks.

My Versions are:
Motioneye: 0.35.2
Motion: 4.0.1

Ringostar1982, 2017-07-17:
I noticed that it doesnt happen now with a resolution of **1280x720**
Motioneye: 0.36.1
Motion: 4.0.1

degustator, 2017-08-17:
Could be that its Infra Red  mode effect, I have the same when IR LED is working with Raspi NOIR camera

Ringostar1982, 2017-08-22:
I have the normal cam not a NOIR camera...

wb666greene, 2017-09-21:
Is auto brightness turned on?

I think it defaults to off and I saw this sometime when restarting after "apply".  You generally get bogus motion detection as the auto brightness works, but the image quality is generally better.

Ringostar1982, 2017-09-25:
No auto brightness is off!  If i turn it on i will get ongoing bogus motion detections and that sucks.

With my latest setup and a less Resolution it works:

Resolution of 1280x720
Motioneye: 0.36.1
Motion: 4.0.1

bobblkabb, 2018-03-13:
Same here with version 0.38. As soon as I save any setting picture turns 'normal' again.

dword1511, 2018-04-01:
Same problem here. Everything below 1024x768 (4:3) and 1280x720 (16:9) are fine, once above all green and overexposed. Does not seem to be related to either width or height alone.

```
$ uname -a
Linux pi3 4.14.30-v7+ #1102 SMP Mon Mar 26 16:45:49 BST 2018 armv7l GNU/Linux
$ dpkg --list | grep motion
ii  motion                         4.0-1                            armhf        V4L capture program supporting motion detection
ii  motioneye                      0.38.1-dmo1                      all          web-based frontend for motion
```

bobblkabb, 2018-04-05:
[image]

swan33, 2018-04-19:
I also gets the same problem... All programs are up to date :(

bobblkabb, 2018-08-10:
Once you change anything and press the apply button everything's getting normal again.

MalFrance, 2018-08-11:
I had the same problem and found elsewhere adding the following 2 lines to the "extra motion options" did the trick.
---
mmalcam_control_params -ex night -awb incandescent -co 10 -sa 20
mmalcam_name vc.ril.camera
---
Apply the changes, wait a few seconds for the camera to re-initialise (or whatever) and works fine (for me).
I can't tell you what these line do precisely, but can guess a little.

Mal

bobblkabb, 2018-08-14:
What does "name value" mean for the _extra motion options_?

I insert those two lines but only the first one is kept..

MalFrance, 2018-08-14:
Until i rebuilt a camera this weekend with motion 4.1.1 i had no idea.
Now i can see it changes the "camera device" setting to the named value.
But interestingly not for every device - i have another pi using older versions of raspbian, motion and motioneye and the line is still shown and no change.
To be honest, i just used copy and paste...

sfgabe, 2018-09-19:
Thanks for this @MalFrance - was having the same problems, though i had better luck getting the green out with setting awb to tungsten

My extra line in thread-1.conf:
`mmalcam_control_params -ex night -awb tungsten -co 30 -sa 30 -sh 50`

bobblkabb, 2019-07-15:
Used motioneyeos (v 0.40) for while without problems. Now I'm back to motioneye 0.40 on dietpi and I'm having greenish pictures again.
The settings are the same I had on motioneyeOS.

Ringostar1982, 2019-10-24:
> Thanks for this @MalFrance - was having the same problems, though i had better luck getting the green out with setting awb to tungsten
> 
> My extra line in thread-1.conf:
> `mmalcam_control_params -ex night -awb tungsten -co 30 -sa 30 -sh 50`

@sfgabe : did this line solved the problem?
 Where must i insert this line exactly ?
thx

JamBalaya56562, 2026-11-08:
Closing, as this was with the Raspberry Pi camera through MMAL, the legacy camera stack, which has been removed from Raspberry Pi OS since Bookworm. The workaround found above was setting the white balance with `mmalcam_control_params` under Extra Motion Options. On current Raspberry Pi OS the camera goes through libcamera, see [Troubleshooting](https://github.com/motioneye-project/motioneye/wiki/Troubleshooting#libcamera-camera-stack). Please open a new issue if pictures still turn green with it.

closed 2026-11-08 by JamBalaya56562