motioneye-project/motioneye #579: motioneye on tinkerboard 4k issue

valdice14, 2017-07-06:
Hi there, sorry in advance for my bad english.
I got work for months motioneye on raspberry pi3 with a 4k cam, i must decrease resolution to 1080p cuz the gpu cant manage the stress.
Now i make a full working brandnew tinker board installing lubuntu 16.04.2, everything run faster then 10 times but when i switch to 4k resolution after refreshing the streaming i got the picture with error without any streaming video.
Any ideas?
Thanks in advance

ccrisan, 2017-07-07:
You'll need to check with [motion](https://github.com/Motion-Project/motion) guys to make sure 4k is currently supported by motion.

closed 2017-07-07 by ccrisan

valdice14, 2017-07-07:
Hemm, like i said with rasperry pi3 and motioneye it work

ccrisan, 2017-07-07:
Still this is not something related to motionEye, which is just a frontend for motion. You need to talk to motion guys and see if there's anything particular about the motion version you use and your board (architecture).

valdice14, 2017-07-07:
Thank you ccrisan

titanity, 2019-01-30:
valdice14 what motioneye image file do you use for tinkerboard? I cant seem to find any Motioneye os for the asus tinkerboard

valdice14, 2019-01-30:
this may help?

ccrisan, 2019-01-30:
@titanity this issue was about motionEye, not motionEyeOS. However we're currently baking a Tinkerboard motionEyeOS image which will be ready for testing tomorrow.

titanity, 2019-01-30:
@ccrisan  
Oh i understand. I can help test if needed Calin. I got a Tinkerboard. Rock64 (4gb). LattePanda Alpha 800. RPI 3bplus. RPI 3b. Odroid XU4.

Also got a normal RPI cam. But also a cam with nightvision aswell... I use both USB as a storage but also my google drive for uploads.

So let me know if i can help with anything at all. 
Btw had a issue with my lattepanda. Had a short circuit and now it wont power up... Bye bye brand new LP Alpha = 500$ . But it runs win10 anyways... So wouldent do much help... Thanks for the fast and constructive reply