motioneye-project/motioneye #580: Email setting getting lost on email disable

ChriZathens, 2017-07-07:
Hello!
In motion notifications, In Send an Email, I wanted to enter necessary settings, so that they are ready in case I want to enable them before e.g. leaving for the weekend.
But I noticed that if you set Send an Email to off again, all settings are getting lost.
I think it makes more sense to allow settings to be remembered, so that it will be a matter of on/apply, or off/apply whenever you want to enable/disable email notifications.
By the current design, every time you want emails you have to configure all settings again.
What do you think?

ccrisan, 2017-07-07:
Given the way those options are implemented (i.e. their presence in the config file means that they are enabled), this would require some serious changes to be implemented.

closed 2017-07-07 by ccrisan

ChriZathens, 2017-07-07:
Yeap... I realize that ...
No big deal - just thought I should mention it as a "nice to have"
Cheers!