motioneye-project/motioneye #582: thread-2.conf over-ridden

carryonrewardless, 2017-07-09:
I have motioneye 4.0.1 installed using pip install motioneye on my raspberry pi 2. When I added a second camera of a similar type as the first the thread-2.conf file does not include the necessary "rtsp_uses_tcp off" line like the first camera does in thread-1.conf.
If I add the line manually it gets over-ridden when the camera is refreshed using the Video Device On/Off slider.
/etc/motion/motion.conf includes the line and I tried adding it manually to the end of /etc/motioneye/motioneye.conf with no improvement.
The cameras are Digoo type and use rtsp://user:pwd@192.168.1.111:554/onvif2 format in configuration.
Thanks for any suggestions.

chrisf4lc0n, 2017-07-09:
I believe I have had a similar conversation with Calin about that... Every time you do something in WebUI the thread-*.conf gets overwritten and there is no workaround that at the moment.

chrisf4lc0n, 2017-07-09:
Try removing the camera and adding it again.

carryonrewardless, 2017-07-09:
Thanks I will try removing and adding again but the other question is why the thread-2.conf is not inheriting the rtsp_uses_tcp off line from the motioneye.conf or motion.conf file. It was a while ago now but I am pretty sure I did not have to add the line to the thread-1.conf for the first camera.

ccrisan, 2017-07-10:
Nothing gets inherited from `motioneye.conf` because it is **not** used to configure motion, but motionEye.

The option `rtsp_uses_tcp` as far as I know is thread (camera)-specific and therefore can't be inherited from `motion.conf` either.

However it should not be overwritten by motionEye if you have properly added your camera (i.e. you didn't copy-paste and edit stuff). If you think this issue is reproducible, please attach your config file(s) and help us reproduce the issue by describing the steps.

carryonrewardless, 2017-07-10:
I deleted and added the camera and it is all working fine. I think I must have not selected the RTSP/UDP pull down option the first time as it has correctly inserted the rtsp_uses_tcp off line.
Thanks for your patience.

ccrisan, 2017-07-10:
Then the issue can be closed.

closed 2017-07-10 by ccrisan