motioneye-project/motioneye #589: Editable mask: which area is detected?

spxak1, 2017-07-16:
This may sound dumb, but when I produce the mask, I cover the image with a set of greyed out boxes and leave parts clear.
Which is then the area detected? The greyed out or the one left clear?

Sorry.

Jabrahx, 2017-07-16:
The one which is grey/white will be detected. But just check it out? :)

ccrisan, 2017-07-16:
The more opaque (dark) regions are *excluded* from the motion detection.

closed 2017-07-16 by ccrisan

jarrodbell, 2018-11-12:
Would be nice to make this more obvious rather than wasting time testing out which is which or searching for an answer. A red cross through the blocks that are ignored, or at least some hint alongside the setting to know which is which.

jeffehobbs, 2019-05-12:
A note in the ? tooltip regarding this color's meaning ("Red blocks are excluded from detection") would be a useful addition.

hobbitlap, 2019-06-01:
and it does not seams to work very well.
if a leave only a small portion (like 10 boxes out of 200), i never get any recordings.
but if a remove the mask option it works great...

gnkarn, 2019-06-08:
in my experience the mask does not work at all , motion is detected anyware in the view window regardless of the mask set .

ntropia2, 2019-08-01:
> in my experience the mask does not work at all , motion is detected anyware in the view window regardless of the mask set .

Same here. I've done some tests masking one half of the picture, and it didn't work. What would be the best way to debug this?

jasaw, 2019-08-05:
If you mask most of your camera view, you'll have to adjust the motion threshold accordingly. If you have 1280x720 pixels (921600 pixels), and masked 90% of your view, and have motion threshold at 11%, it will never trigger motion detection.

Mask has been working great for me. I suggest you enable the mask and turn on "show motion frames" to work out the appropriate threshold.

jurgenweber, 2020-03-08:
ok, so the red covered areas are ignored? Finding it really hard to get successful results, I have the detection at 1% so it should record pretty much when anything moves. I might hvae 20% of the photo covered in red.

zajc3w, 2020-03-13:
motion detection detect motion on perfectly still image with 50% masked and threshold set to 10%...

JacquesBB, 2020-05-25:
It is just impossible to remember which part is excluded. This should be added in the online doc ( with the ? mark).

BETLOG, 2020-11-11:
I came here looking to confirm that red is masked OUT. So I'll confirm that logically it does make sense that way.

I'd also like to suggest that 'frame change threshold' should be the value remaining AFTER  the mask is accounted for. Not a percentage of the total resolution.

gyurobenjamin, 2020-12-27:
Overall documentation needs to be improved. Almost each setting needs more info, but this was a major issue. I spent a day on this trying to debug what can be wrong. Even with red colour, I was expecting to select the active area.

BETLOG, 2020-12-28:
> each setting needs more info

Please don't be 'that developer' who makes the help tip innane. 
eg: File save (saves file)
Be the developer who puts huge swathes of text that is actually useful on mouseover help tips.
Theres a lot of mouseover tip buttons that could easily contain much more useful text, but don't.

leonardochaia, 2022-11-11:
>  ("Red blocks are excluded from detection") 

That would have saved me two days of recording cars moving instead of my door 

> Even with red colour, I was expecting to select the active area.

That was my reasoning as well.
Again, tooltip would have helped a lot.

[image]

alexandervlpl, 2024-09-09:
I've been here like 5 times to check this, with any motion detection software it seems to be a 50/50 chance which way it's done. :man_facepalming: