motioneye-project/motioneye #595: gdrive uploads

MalFrance, 2017-07-20:
Hi,

My first foray into github, so hope this is the right way to go about things.

Motioneye installed and working onto a Pi3b using the latest build of Jessie and a single Pi camera module. No problems in itself.
Great package by the way, so much better than my own homebrew attempt. :)

Gdrive is uploading, that itself is no major problem.
However, the gdrive uploads/syncs seem to come at irregular times. Sometimes _seems_ like it takes an hour or so before a picture is uploaded, sometime just minutes.
What is the planned schedule or interval to upload images to gdrive.
Is there a way to manage it, or at least check when the uploads are occurring?

secondly, i set the path to upload as /var/lib/motioneye, as this is where the pictures are stored.
Yet, rather than just uploading the ./motioneye/ directory itself, the whole path of /var/lib/motioneye/ is uploaded.
Is there any way to restrict the uploads to just the contents of the ./motioneye/ directory?
Or, some way to add the camera name inside the motioneye directory to differentiate which camera is which?
[image]
[image]

As a slightly different point, i can't find any motioneye.log anywhere in the system (find from root doesn't find it)
How can i activate motioneye logging? 

Cheers,

Mal

cybercorey, 2018-04-16:
Hi Mal,
For your path issues, just set the localtion to /motioneye (this is the path in gdrive only)
As for your other issue, I've having the same problem, heres another ticket for that one: #833

JamBalaya56562, 2026-11-07:
Closing as answered. Uploads are not scheduled: each picture or movie is queued for upload about five seconds after Motion saves it, so long delays point to the upload itself, which `motioneye.log` shows. The location is the folder on Google Drive, as cybercorey said above, not the local path; "Include Subfolders" decides whether the local subfolders are kept under it. Please open a new issue if uploads still lag with a current release.

closed 2026-11-07 by JamBalaya56562