motioneye-project/motioneye #596: 401 Unauthorized

bruffdj, 2017-07-20:
Hi,

I've been happily using motion eye os on a raspberry pi with my wansview camera for some time now.  

I now need to switch over to running motion eye on Ubuntu.  When I try to connect with the same URL, username and password it does not work, and the last line in motion.log is Server returned 401 Unauthorized.  If I disable authentication on the camera's RTSP settings I can get connected from.  I don't want to leave authentication disabled.  

Other tests I've done:

1) I created a new user on the camera with a very simple username and password, I can connect to that fine from the pi, but experience the same problem from the Ubuntu setup.  

2) I can connect to the camera from ubuntu using VLC whether authentication is disabled, or using either of the account available on the camera.  

It seems to me that the Ubuntu setup is not passing the credential correctly.  

Can you help?

Thanks

George

c-geek, 2017-08-31:
I encountered exactly the same problem today. I had a Raspberry PI working fine with MotionEye and this camera. But then I wanted MotionEye on an Ubuntu PC.

My Ubuntu was a 14.04 just upgraded to 16.04. The installed version of motion was 3.2.11. That was the issue.

Upgrade to Motion@4.0.1-1 fixed the issue: https://github.com/Motion-Project/motion/releases

```
wget https://github.com/Motion-Project/motion/releases/download/release-4.0.1/xenial_motion_4.0.1-1_amd64.deb
sudo dpkg -i xenial_motion_4.0.1-1_amd64.deb
sudo systemctl restart motioneye
```

ccrisan, 2017-09-01:
@bruffdj what version do you use? Like @c-geek said, if you're on 3.x, please upgrade to 4.x.

bruffdj, 2017-09-01:
Awesome, thanks @c-geek! Problem solved, little annoyed with my self for not thinking to check that! Thanks for sharing.

closed 2017-09-01 by ccrisan