motioneye-project/motioneye #600: Scricam Invalid data found when processing input

codingyourlife, 2017-07-23:
Hi. I have two cameras which I know have rtsp because ONVIF Device Manager shows the stream and I used shinobi before which worked too but I had other issues and wanted to try motioneye. With shinobi I had troubles too to get the cams running and my issue here helped ([issue](https://github.com/moeiscool/Shinobi/issues/114))

**motion.log**
[1:ml1] [WRN] [NET] netcam_interrupt_rtsp: Camera timed out for rtsp://192.168.178.67:554/onvif1
[1:ml1] [ERR] [NET] netcam_rtsp_open_context: unable to open input(rtsp://192.168.178.67:554/onvif1): Invalid data found when processing input
[1:ml1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.178.67:554/onvif1)
[1:ml1] [ERR] [ENC] ffmpeg_avcodec_log: Nonmatching transport in server reply
[1:ml1] [ERR] [NET] netcam_rtsp_open_context: unable to open input(rtsp://192.168.178.67:554/onvif1): Invalid data found when processing input
[1:ml1] [WRN] [ALL] motion_init: Could not fetch initial image from camera Motion continues using width and height from config file(s)

**motioneye.log**
2017-07-23 22:56:21: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2017-07-23 22:56:31: [motioneye]    ERROR: connection problem detected for mjpg client for camera 1 on port 8081
2017-07-23 22:56:31: [motioneye]  WARNING: Connect error on fd 14: ECONNREFUSED
2017-07-23 22:56:31: [motioneye]  WARNING: Connect error on fd 14: ECONNREFUSED
2017-07-23 22:56:31: [motioneye]  WARNING: Connect error on fd 14: ECONNREFUSED
2017-07-23 22:56:31: [motioneye]  WARNING: Connect error on fd 14: ECONNREFUSED
2017-07-23 22:56:31: [motioneye]  WARNING: Connect error on fd 14: ECONNREFUSED
2017-07-23 22:56:31: [motioneye]  WARNING: Connect error on fd 14: ECONNREFUSED
2017-07-23 22:56:51: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2017-07-23 22:57:01: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2017-07-23 22:57:11: [motioneye]    ERROR: mjpg client timed out receiving data for camera 2 on port 8082

[image]
[image]

ccrisan, 2017-07-24:
You'll need to supply the correct RTSP URL. The URL you supplied is clearly not correct. You might need to look online for the RTSP URL of your camera.

closed 2017-07-24 by ccrisan

codingyourlife, 2017-07-24:
The URL should be fine. See here: http://www.sricam.com/article/id/7420c0f522b14d63898878b0581e80df.html
(last picture)

ccrisan, 2017-07-25:
If you really think the URL is correct, please open an issue on the [motion](https://github.com/Motion-Project/motion) project, providing all required information (logs, config files).

komalpharate, 2018-01-26:
Hi,
Is it like Sricam's rtsp://<ip_address>/onvif1 this RTSP+onvif1 URL's behaviour is diffrent from normal RTSP urls.
Because I'm able to open and get frames from the rtsp urls but not with this onvif1 url.

kimthostrup, 2018-12-06:
Thanks for your reply @gusarg81 but my unbranded China cam does not support udp :-/