motioneye-project/motioneye #601: empty HTTP respons when curl'ing snapshot URL

MrMontesa, 2017-07-24:
In #94 I came to know that it is possible to curl images from motioneye and process them further. I also recognized that sometimes curl is not able to download the image but HTTP/200 OK was sent.
As a temporary workaround I added multiple curl commands in my script until there is data transmitted.

Unfortunately this does not work with a 3rd party (Openhab2) app I'm using. I also created a ticket with OH2 dev team and we analyzed the issue in detail from OH2 side of things. After some tcpdumps we came t know that motioneye does not seem to send the image but sends OK - 200.
You can read some more details in the relevant [github issue](https://github.com/openhab/openhab1-addons/issues/5254)

Now I'd like to focus in the motioneye implementation. Is there any way to troubleshoot this in more detail?
Thanks for your help.

ccrisan, 2017-07-24:
@MrMontesa what version of motionEye were you running when you got HTTP/200 OK with no image?

MrMontesa, 2017-07-25:
I'm sorry to not have confirmed my environment. Here are some details:

Hardware: Pi Zero W
Cam: NoIR2
OS: Debian 8
App: 

```
swa@pizero:~ $ /usr/bin/python /usr/local/bin/meyectl -v
motionEye 0.35.2
swa@pizero:~ $ /usr/bin/python --version
Python 2.7.9
```
Let me know if I can be of any help

ccrisan, 2017-07-25:
Please upgrade to 0.36.x.

closed 2017-07-25 by ccrisan

MrMontesa, 2017-07-25:
Thanks for info. Upgrading to 0.36.1 seems to properly fix the issue.
Would be good to also have the release data available on the [changelog](https://github.com/ccrisan/motioneye/wiki/Changelog) page.
Again, thanks much for your swift help.

ccrisan, 2017-07-25:
It's there, with 0.36:

    snapshot URL will do its best to always serve a JPEG frame

MrMontesa, 2017-07-25:
Yeah, I'm sorry for the typo. I meant "release date" not data. Like the date you released the version.

ccrisan, 2017-07-25:
Actually what I need is to move the changelog details to the github releases page. Then everything will make more sense and release dates will be visible as well.