motioneye-project/motioneye #602: when using action button - tricker a python script - I get error

guitarjhk, 2017-07-25:
I get "An error occurred. Refreshing is reccommended" - the action button is placed in /etc/motioneye/ and call "left_1" and it call "/usr/bin/python3 /etc/motioneye/left.py" - if I look in /var/log/motion.log there is no error -  if I change log to debug - I still don't get any detial about that error - If I run the the call from a "left.sh" script it works fine 
the action button file is executable and has no extension 

[image]

ccrisan, 2017-07-25:
Please post the contents of your left_1 file.

guitarjhk, 2017-07-25:
root@pantiltrsp:/etc/motioneye# more ./left_1
/usr/bin/python3 /etc/motioneye/left.py
root@pantiltrsp:/etc/motioneye# more ./left.py
#!/usr/bin/python
import time
import pantilthat

#---------------------

#---------------------

pantilthat.servo_two(-10)
time.sleep(1)