motioneye-project/motioneye #603: Impossible to send mails notifications

Kywalh, 2017-07-27:
Hi all,
I'm trying to make the mail notification working but I keep on having the connection timeout error...

here is the log:

> [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
[0:motion] [WRN] [ALL] thread config option deprecated use camera
[0:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
[0:motion] [NTC] [ALL] motion_startup: Motion 4.0.1 Started
[0:motion] [NTC] [ALL] motion_startup: Logging to syslog
[0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1:ml1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (http://xxx.xxx.xxx.xxx:xxxx/videostream.cgi?user=*******&pwd=********)
[1:nc1] [ALR] [NET] netcam_handler_loop: Camera handler thread [3] started
[1:ml1] [WRN] [NET] netcam_next: called with no data in buffer
[1:ml1] [WRN] [NET] netcam_next: called with no data in buffer
2017-07-27 16:33:31: [sendmail]    ERROR: failed to send mail: Connection unexpectedly closed
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/motioneye/sendmail.py", line 210, in on_message
    options.tls, _from, to, subject, message, files or [])
  File "/usr/local/lib/python2.7/dist-packages/motioneye/sendmail.py", line 53, in send_mail
    conn = smtplib.SMTP(server, port, timeout=settings.SMTP_TIMEOUT)
  File "/usr/lib/python2.7/smtplib.py", line 256, in __init__
    (code, msg) = self.connect(host, port)
  File "/usr/lib/python2.7/smtplib.py", line 317, in connect
    (code, msg) = self.getreply()
  File "/usr/lib/python2.7/smtplib.py", line 368, in getreply
    raise SMTPServerDisconnected("Connection unexpectedly closed")
SMTPServerDisconnected: Connection unexpectedly closed
2017-07-27 16:34:52: [sendmail]    ERROR: failed to send mail: Connection unexpectedly closed
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/motioneye/sendmail.py", line 210, in on_message
    options.tls, _from, to, subject, message, files or [])
  File "/usr/local/lib/python2.7/dist-packages/motioneye/sendmail.py", line 53, in send_mail
    conn = smtplib.SMTP(server, port, timeout=settings.SMTP_TIMEOUT)
  File "/usr/lib/python2.7/smtplib.py", line 256, in __init__
    (code, msg) = self.connect(host, port)
  File "/usr/lib/python2.7/smtplib.py", line 317, in connect
    (code, msg) = self.getreply()
  File "/usr/lib/python2.7/smtplib.py", line 368, in getreply
    raise SMTPServerDisconnected("Connection unexpectedly closed")
SMTPServerDisconnected: Connection unexpectedly closed
curl: (28) Operation timed out after 5000 milliseconds with 0 bytes received
curl: (28) Operation timed out after 5001 milliseconds with 0 bytes received
2017-07-27 16:36:52: [sendmail]    ERROR: failed to send mail: Connection unexpectedly closed
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/motioneye/sendmail.py", line 210, in on_message
    options.tls, _from, to, subject, message, files or [])
  File "/usr/local/lib/python2.7/dist-packages/motioneye/sendmail.py", line 53, in send_mail
    conn = smtplib.SMTP(server, port, timeout=settings.SMTP_TIMEOUT)
  File "/usr/lib/python2.7/smtplib.py", line 256, in __init__
    (code, msg) = self.connect(host, port)
  File "/usr/lib/python2.7/smtplib.py", line 317, in connect
    (code, msg) = self.getreply()
  File "/usr/lib/python2.7/smtplib.py", line 368, in getreply
    raise SMTPServerDisconnected("Connection unexpectedly closed")
SMTPServerDisconnected: Connection unexpectedly closed

> 
and here is the configuration:

[image]

and the error message:

[image]

On the gmail side, I activated the "authorize vulnerable applications" or something like that after I received notifications from gmail that something was trying to connect to my account..

I tried also with my yahoo account but Motioneye doesn't support SSL certificate...

HELP

rflagler, 2017-08-23:
I think smtp.gmail.com uses port 587. My config works just fine that way.

divemasterjm, 2020-04-15:
@lwvining how do you use a text messgae in the email? i want to send camera name, just to trigger some options in another system

lwvining, 2020-04-15:
In the setting I use my phone number@vzwpix.com for Verizon. For other cell providers, of course, the @xxxx.xxx is different. You can google that. 
Since I use gmail, I’m using my gmail address in the SMTP settings. Be sure to use the correct port for gmail (587). Also for security you may need to register your email app use with a security password for the app on you account.
I attached two screenshots for your reference. I have two cameras on this pi and one stopped working as you can see from the shot , it’s black.Also the text message screen shot is from a different camera setup. Hope that’s not confusing.

Not sure if this is what you want. My setup sends me a text on motion trigger. The normal setup is for sending an email on motion triggers . I think this answers your question. Good luck.
Regards,
Lynnette

JamBalaya56562, 2026-11-08:
Closing, as this was with motionEye on Python 2 in 2017. "Connection unexpectedly closed" came while connecting to the SMTP server, which happens when the port and the TLS setting don't match the server. For Gmail, port 587 with TLS worked for others above, and Gmail needs an app password as the SMTP password. Please open a new issue with the `motioneye.log` lines if mail still fails with a current release.

closed 2026-11-08 by JamBalaya56562