motioneye-project/motioneye #604: Motion being run in debug mode: large log files consume SD card storage

kenmoore, 2017-07-27:
I leave motioneye running continuously, but every once in a while I am having to delete large user.log and debug log files from /var/log directory -- they appear to be full of log messages like those pasted below.  Is there some setting that I'm overlooking?  At one point I had enabled "Create Debug Media Files" but have since disabled that option as well as the entire Motion Detection group, so it seems no debug logging should be created.

Dec 31 16:33:45 pi2 motion: [1:ml1] [DBG] [VID] v4l2_next: 1) vid_source->pframe 3
Dec 31 16:33:45 pi2 motion: [1:ml1] [DBG] [VID] v4l2_next: 2) vid_source->pframe 3
Dec 31 16:33:45 pi2 motion: [1:ml1] [DBG] [VID] v4l2_next: 3) vid_source->pframe 0 vid_source->buf.index 0
Dec 31 16:33:45 pi2 motion: [1:ml1] [DBG] [VID] v4l2_next: vid_source->buf.bytesused 720000
Dec 31 16:33:45 pi2 motion: [1:ml1] [DBG] [VID] v4l2_next: the_buffer index 0 Address (7384d000)
Dec 31 16:33:45 pi2 motion: [1:ml1] [DBG] [VID] v4l2_next: 1) vid_source->pframe 0

ccrisan, 2017-07-27:
Is this motionEye or motionEyeOS?

kenmoore, 2017-07-29:
I'm using motionEye.

ccrisan, 2017-07-30:
What's your log level setting in your motioneye.conf file?

kenmoore, 2017-07-30:
Ok, it was set to debug... I'm guessing I did that myself and duly forgot!  I've set it to warning -- this will report errors and warnings but not info or debug messages, correct?  Thanks

ccrisan, 2017-07-31:
Yeap, you should set the log level to info.

closed 2017-07-31 by ccrisan