motioneye-project/motioneye #605: Motion trigger never ending

pbo10, 2017-07-29:
Hi All

I found an issue in Motioneye where if you're using video recording set to continuous, the motion trigger never actually ends. This means any motion notifications you have set up never trigger after the first time.

My reason for wanting this is to have a continuous 24/7 recording, but use the motion detection feature with a mask so I can use the web hook to trigger external events if there's motion in a specific area on camera. The motion detection still works perfectly and highlights the areas of movement, but it seems that as soon as you enable the continuous recording it affects the motion event ending in some way because even with no motion detection at all for extended periods it stays in the triggered state.

Is there any way to fix this at all?

Thanks!

ccrisan, 2017-07-30:
I don't think there is a simple solution to this. If however anyone finds a way to have continuous recording while events work normally, please let me know and I'll see what I can do.

closed 2017-07-30 by ccrisan

sbourdette, 2017-12-29:
It's working with motion with the folowwing config :

`
# Use ffmpeg to make videos showing the moving pixels (ghost images) (default: off)
ffmpeg_output_debug_movies on

# Bitrate to be used by the ffmpeg encoder (default: 400000)
# This option is ignored if ffmpeg_variable_bitrate is not 0 (disabled)
ffmpeg_bps 400000

# Enables and defines variable bitrate for the ffmpeg encoder.
# ffmpeg_bps is ignored if variable bitrate is enabled.
# Valid values: 0 (default) = fixed bitrate defined by ffmpeg_bps,
# or the range 1 - 100 where 1 means worst quality and 100 is best.
ffmpeg_variable_bitrate 75

# Container/Codec output videos
# Valid values: mpeg4, msmpeg4, swf,flv, ffv1, mov, mp4, mkv, hevc
ffmpeg_video_codec mp4

# When creating videos, should frames be duplicated in order
# to keep up with the requested frames per second
# (default: true)
ffmpeg_duplicate_frames true

# Interval in seconds between timelapse captures.  Default: 0 = off
timelapse_interval 1

# Timelapse file rollover mode. See motion_guide.html for options and uses.
timelapse_mode hourly

# Frame rate for timelapse playback
timelapse_fps 4

# Container/Codec for timelapse video. Valid values: mpg or mpeg4
timelapse_codec mpeg4

`