motioneye-project/motioneye #608: Unit motioneye.service entered failed state.

bigfoot50, 2017-07-31:
Hi,

I installed motioneye on a openmediavault machine (Debian 8) like in your wiki wrote. Everything went smooth, but when I start the service nothing happens. 

with journactl -xn
```
working_set = WorkingSet._build_master()
Jul 31 18:17:53 openmediavault meyectl[3486]: File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 449, 
Jul 31 18:17:53 openmediavault meyectl[3486]: ws.require(__requires__)
Jul 31 18:17:53 openmediavault meyectl[3486]: File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 745, 
Jul 31 18:17:53 openmediavault meyectl[3486]: needed = self.resolve(parse_requirements(requirements))
Jul 31 18:17:53 openmediavault meyectl[3486]: File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 639, 
Jul 31 18:17:53 openmediavault meyectl[3486]: raise DistributionNotFound(req)
Jul 31 18:17:53 openmediavault meyectl[3486]: pkg_resources.DistributionNotFound: MarkupSafe>=0.23
Jul 31 18:17:53 openmediavault systemd[1]: motioneye.service: main process exited, code=exited, status=1/FAILURE
Jul 31 18:17:53 openmediavault systemd[1]: Unit motioneye.service entered failed state.
```
On my raspbery pi3 it worked perfect. 

Installed are phyton3 and 2.7. It seems to be really hard to install a second service on an openmediavault machine.

Any idea? 

Regards from Germany

chrisf4lc0n, 2017-08-03:
Can't you just install plain Rapbian and install all the services run on OMV plus MotionEye?

Ohem, 2017-08-03:
Already tried to but OMV demands php5 and some dependencies to be installed and for some reason unable to retrieve. I also tried with OMV image preinstalled and then install motioneye on top - but pycurl and pillow gives me errors.

Had to give up on OMV and turn to manual mount, permissions and automount upon reboot. Like in this guide: https://www.modmypi.com/raspberry-pi/accessories/hard-drives/seagate-1tb-2.5-external-hard-drive-rpi-tested/?search=external%20drive as I'm only interested in one drive I jsut skip the backup part.

JamBalaya56562, 2026-11-07:
Closing, as this was the Python 2 release on Debian 8: `meyectl` stopped because the MarkupSafe found was older than the 0.23 that motionEye needed. Current releases run on Python 3 and are installed with pip, which brings the dependencies motionEye needs, see the [README](https://github.com/motioneye-project/motioneye#installation). Please open a new issue if the service still fails to start with a current release.

closed 2026-11-07 by JamBalaya56562